FreeCAD/FreeCAD #9660: Revolution/Mirroring produces a gap in a body

holmas, 2023-05-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.0.0
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * CurvedShapes 1.0.4
[/code]
```

### Subproject(s) affected?

None

### Problem description

While building the attached body a gap was produced as shown in picture #2. I had noticed this gap when I wanted to make fillets on the edges shown in picture #1. After reloading of the file the gap was gone, but it is still impossible to make that fillets.

### Anything else?

[fehler_entriegelung.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11564381/fehler_entriegelung.FCStd.zip)
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-07:
Heads up. please don't use `#1` or any number after a hashtag because it will link to another ticket in github. I've fixed this in the original ticket for you by adding Nº1 and Nº2. TIA!

luzpaz, 2023-06-07:
Is this then just UX error and if so do we do anything about it ?

grubertechnic, 2023-06-08:
It is a mixture of coplanar issue and user error. I would close it.

grubertechnic, 2023-06-29:
I would close it. It is a user error. And for the coplanar issue (which is in that case not really the source of the issue), there are dozens of duplicates in the tracker.

grubertechnic, 2023-06-30:
@adrianinsaval or @luzpaz : i definitely think, that is a candidate to close it.

closed 2023-06-30 by adrianinsaval