FreeCAD/FreeCAD #9666: [Problem] Boolean op fails for tap and die from fastener workbench

holmas, 2023-05-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.0.0
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * CurvedShapes 1.0.4
[/code]
```

### Subproject(s) affected?

None

### Problem description

Making a boolean difference fails as follows:

a) difference between a box and a "ScrewTap" (Fasteners workbench) [picture 1]
It's hard to see in the picture, but all vertical faces inside the hole are gone, so this isn't a solid body anymore.

b) difference between a rod and a "ScrewDie" (...dito...) [picture 2]
The vertical surfaces should not be present here. On the other hand, all other surfaces of the thread are missing.

The provided test project contains both cases.

### Anything else?

[image]
[image]
[test.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11567233/test.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-05-28:
2 things need to happen: 
1. reproduce this on v0.21 dev version 
2. let the fastener workbench (https://github.com/shaise/FreeCAD_FastenersWB) dev know

grubertechnic, 2023-06-30:
Same as #9751, co-planar touching.

luzpaz, 2023-07-01:
@grubertechnic you're saying this a duplicate (and not specifically related to fasteners workbench) ?

CC @shaise

grubertechnic, 2023-07-02:
> you're saying this a duplicate (and not specifically related to fasteners workbench) ?

The principal the same (co-touching in general), but the situation is finer to resolve.

While the other issue is about
-  b-spline co-touching (b-spline nears circle of and floating numbers) on both sides

this here is about
 - co-touching b-spline (b-spline nears circle of and floating numbers) and on the other side an exact cylinder with float number (almost neglectable).

It can work, but must not, since the co-touching is fuzzy. See here on the 3rd picture, it is suggest to overlap, https://wiki.freecad.org/Thread_for_Screw_Tutorial#Tips_for_success.

Side note, also commercial programs have rarely (with clarification of rarely) issues with that. Not that often, but still, and they will not accept any ticket without an overlap. That issue is intrinsic due its conception of boolean operation with due numerical help, you can near it better and better, but you will never get 100% rid of it.

luzpaz, 2023-09-05:
CC @shaise

luzpaz, 2024-03-09:
cc @shaise

shaise, 2024-03-10:
Hi @luzpaz ,
Sorry for missing that. 
I'm not sure its Fasteners WB related. I think OCCT is just having problems sometimes doing boolean operations on complex models (such as threads). Whenever I encounter such an issue I try to play with slight "epsilon" changes to dimensions until the boolean works.

luzpaz, 2024-03-14:
@maxwxyz any suggestion on what we should do here ? Link to known OCC behavior and close or keep open but rename?

maxwxyz, 2024-03-14:
@luzpaz I guess if anytime resolved in OCC this would be no issue anymore? Is this also the case for the linked https://github.com/FreeCAD/FreeCAD/issues/9751 ?
Is there an existing issue on the OCC repo? If so we could link it here and close this.

krushia, 2024-07-25:
I can't reproduce this in yesterday's 0.22 source with OCC 7.6.3. Is this really an OCC issue? If so, was a workaround introduced and not mentioned here?

Also seems fixed in 0.21.2, but it requires selecting every object, marking for recompute, then recomputing document.

```
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
```

Also fixed in AppImage built with OCC 7.7.2
```
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

TL;DR: Issue resolved, but not sure how it was resolved.

maxwxyz, 2024-07-25:
@holmas can you confirm?

AIRCAP, 2024-10-17:
the missing vertical faces are fixed in latest main (Cut001) but the broken surface still occurs (Cut)

fixed in  #17119  though.

PR #17119 (2024-10-21): Fix boolean operations
Fix #17159
Fix #17150
Fix #17085
Fix #16081 
Fix #15825
Fix #15615
can alleviate #11083 ( fixes with nonstandard setting for boolean fuzzy > 1000 - but that's too high to cut out very delicate parts from very large blocks )
Fix #9666
Allows workaround for #17338 

OCCT has issues with boolean operations failing when parts are touching but not intersecting. This includes but is not limited to the PartDesign helix , which is notorious for producing results that do not pass BOP tests or even fail completely.

The suspected problem is too narrow precision. Part precision sho [... cut]
closed 2024-10-21 by yorikvanhavre