FreeCAD/FreeCAD #9671: [Problem]  Part Design error in boolean operation (fuse)

siern215, 2023-05-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.56
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Problem description

the problem is described in the following images
[image]
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-26:
please upload the file

stern85, 2023-05-29:
[Prova 4.zip](https://github.com/FreeCAD/FreeCAD/files/11591238/Prova.4.zip)

adrianinsaval, 2023-05-30:
unfortunately that is expected behavior, PD boolean operations move objects to have their position relative to the body's origin system, you'll notice in your example for the body that "works" if you move that body before making the boolean OP the same happens. The PD boolean OPs need a rework to be actually useful

grubertechnic, 2023-06-04:
> unfortunately that is expected behavior

Fortunately that is expected behavior, this is an absolute must! Else every transform of the main Body will destroy the result. It is inherent logic.

grubertechnic, 2023-06-29:
Is it really needed to keep that open? It works as expected.

adrianinsaval, 2023-06-29:
re read the title, it might be working as intended but is terribly unintuitive behavior.

> this is an absolute must! Else every transform of the main Body will destroy the result. It is inherent logic.

how come subshapebinders can work even if you move the body then? This is a limitation of the current implementation and it's certainly not simple to solve, but it is not a must.

grubertechnic, 2023-06-29:
> how come subshapebinders can work even if you move the body then?

SubShapeBinder were included later as exception and is only a patch, never need them really and there are for me quite unnatural. Why do they have a quasi hidden 2D plane offset function it it? Why the the make face property on a function, that accepts all kind of objects and a face have no function in PartDesign, since it works with wires anyway? Why all this properties? If the ShapeBinder itself have all this properties and functions, there would be no need for 2 binders types. Nothing against the option as follow mode (or bind mode?, idk) as property, it ___can___ make sense.

Anyway, everything else uses the coordinate system as the main Body.

> it might be working as intended but is terribly unintuitive behavior.

I recognize this behavior as absolute natural, never had issues with it. Also seems that not many have issues with it, since there is not many noise about that. The only thing i can imaging, that there is a property like in the SubShapeBinder for follow mode (or bind mode?).

But the tile and its content _Part Design boolean operations are unintuitive and very limited_ is for me not true and bogus.

Side note: there is a bug with not Bodies solid objects and boolen operation in PartDesign, but that is not the case here.

adrianinsaval, 2023-06-29:
I suspect the reason for the little noise is because they are rarely used and most of the time when used the objects are already both at origin so it doesn't become an issue.

grubertechnic, 2023-06-29:
If then, when someone came with bigger moldings and/or injection moldings, this kind workflow is unavoidable, else the parts gets unmanageable and unmaintainable.

adrianinsaval, 2023-06-29:
I don't understand what you are trying to say, do you not want the placement of your parts to be respected? I found current PD booleans to be unintuitive and buggy exactly while working on an injection mold [model](https://github.com/GliaX/tourniquet/blob/master/injection_molding/Glia%20Tourniquet%20Inj-Mold%20Design%20(FreeCAD)/sliderAssy.FCStd) I had to workaround it with a subshapebinder and then a clone of the subshapebinder to actually make a boolean from the plastic part.
Requiring workarounds for simple tasks like this is unintuitive and counterproductive.

grubertechnic, 2023-06-30:
> I found current PD booleans to be unintuitive and buggy exactly while working on an injection mold [...]

I do not see, where there issue is specific, also there are issues while loading the file. Can you point them (issues)? Just said, normally the build up is different (each slider a part) and the injection part is "dead" object. When you replace the part due specific revision, then a compare tool (which face, line, vertex and their orientation is linked and replace/reorder them) is needed anyway. Even else with correct TNP it will not work anyway without bigger issues.

I mean models like this: https://forum.freecad.org/viewtopic.php?p=680448#p680448, not the tools. There you build Body in Body in different depths. Atm you can not do this in FreeCAD, since the TNP and some boolean operations (trim or selective, or other name) are missing. But the coordinate system is the base for that. With that conception you can manage files with >10000 objects and >1.5GByte for a singe part (no assembly!) pretty good.

(Since i know, where one founder was working and i was working in a component supplier company for them and i had to use the same system too, i know the intend beneath, and it is crazy good, it sounds really awful, but it is smart af).

> I don't understand what you are trying to say, do you not want the placement of your parts to be respected?

Only in chains. Coordinate = WorldCS + AssemlyCS + PartCS + BodyCS + (boolean)BodyCS + (boolean)BodyCS + .....
But here is nothing against a property like the SubShapeBinder like Relative

[image]

Btw, this is not specific here, but the ShapeBinders must be refactored, since two are useless (can be merged) and the "attachment" issue, where i have forgotten to crate a ticket.

https://forum.freecad.org/viewtopic.php?style=10&t=76445

adrianinsaval, 2023-06-30:
> I do not see, where there issue is specific, also there are issues while loading the file. Can you point them (issues)? 

You need to have the linked files available to properly load the file, can't remember now exactly which ones. About the issue well you won't immediately see in the file because I have workaround it, but in the body "slide" remove the boolean operation, the group "Auxiliary" remove Body002 and Binder. Now try to make a boolean cut with the "clip" Link. But I have to say I was wrong to bring this one in this thread since the issue here is with poor support for other object types. The slide is in the origin position there too. But bodies can be translated, so it is not good that boolean OPs can't be made on translated bodies.

> Just said, normally the build up is different (each slider a part) and the injection part is "dead" object.

Even so, can you really tell me that is wrong to position a body/part and then make a boolean? That sounds absurd to me, why a have a placement property in a body if we can't use it?

grubertechnic, 2023-06-30:
> Even so, can you really tell me that is wrong to position a body/part and then make a boolean?

Maybe there is a misunderstanding, i am for changing the position, but it should be relative to the target Body.

A small example, a bearing area of a casting, a liitle bit artificial (i would build it a little bit different, i would make the oil supply from the outside, but it is just a showcase), but i shows the principle ok. Even it is a singe Body in the end, through boolean operations it acts like an assembly in a Body.

A few notes:
 - normally the two Pockets _normally_not_needed_ are not needed, because the boolean operations should be not an union, it should be a boolean trim (or selective boolean, named different on different systems), which is not implemented in FreeCAD.
  - While recording the video, is saw, that i make in the small pocket from the workaround i small mistake, it follows not in every direction, when changing the oil channel.
 - Still a little bit brittle due TNP. But when you have many smaller Bodies, then the TNP behavior is much better, even when the TNP in FreeCAD is solved. Because the TNP on larger models is always a little problem.
 - I change the stylesheet from default to an other, because the default stylesheet the tree view is not very good. It is missing the small lines, which is needed for a better overview.

[2023-06-30 21-09.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/aa35cd6c-1be9-4cd1-872b-29e037554cd7)

[test.zip](https://github.com/FreeCAD/FreeCAD/files/11919896/test.zip)

adrianinsaval, 2023-07-03:
I can see where you are coming from, but if both bodies are already translated and positioned _before_ launching the boolean operation it is very unintuitive that the tool body ends up nowhere near where you put it. And note that if the target body is already positioned there is no way right now for you to put the tool body on the same coordinate system context before launching the boolean operation so that you can properly place it in relation to the body origin (well you could do the math but that's not a friendly workflow), and since boolean OPs can be computationally expensive it is pretty bad if the user has to first make a boolean OP with the part badly positioned to then be able to fix the position to get the desired result (and fixing the position could imply multiple auto recomputes)

It's not clear to me what is the best way to solve this yet, but I think it's a legitimate UX problem, if anything at least an explanatory warning should be displayed.

grubertechnic, 2023-07-04:
> but if both bodies are already translated and positioned _before_ launching the boolean operation it is very unintuitive that the tool body ends up nowhere near where you put it

OK fair point. Maybe the boolean task dialog can be optimized about its transformation and its handling.

maxwxyz, 2024-11-01:
What is left to do here?

UTurer, 2024-11-02:
Sorry I am a newbie but I found this issue while trying to do some boleans. Is this experience expected behaviour? [](https://www.youtube.com/watch?v=OJw9ZHNArz4)

closed 2025-02-22 by github-actions[bot]