FreeCAD/FreeCAD #9672: [Problem] Replace uses of missing font family "FreeCAD NaviCube"

andrewknockin, 2023-05-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 13.3.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.0.0
  * A2plus 0.4.60k
  * kicadStepUpMod 10.17.1
  * toSketch 1.0.1
  * Assembly4 0.12.6
  * Assembly3 0.11.4
[/code]
```

### Subproject(s) affected?

None

### Problem description

The error message in the title appears in the report window:

1. after quite a few seconds (maybe 40) when I first ran this version of FreeCAD
2. on subsequent runs when I open Preferences (almost immediately)
3. on subsequent runs when I open a model file (almost immediately)

The full error message is:
Populating font family aliases took 270 ms. Replace uses of missing font family "FreeCAD NaviCube" with one that exists to avoid this cost.

### Anything else?

https://forum.freecad.org/viewtopic.php?p=684168#p684168

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jolbas, 2023-06-03:
Now when the font size of navigation cube is chosen to fit, maybe we could get rid of all default font code and just use the system font as default. You would of cause still be able to  choose a custom font in the preferences.

adrianinsaval, 2023-06-03:
True! I think getting generic fonts didn't work well on windows though so I think we could set Arial as default, it should be available on windows and mac and get a decent substitute on linux

PR #9724 (2023-06-07): Improved font handling on Navigation cube
Adding anisotropy texture mapping for better readability on steep angles. Tip from aapo in this post https://forum.freecad.org/viewtopic.php?p=657880#p657880
Before:
[image]
After:
[image]

Don't save a font name to parameters if none is chosen in preferences
[image]

Fix #9672 because the "FreeCAD NaviCube" is no longer created.

- [X]  The Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
closed 2023-06-07 by adrianinsaval