FreeCAD/FreeCAD #9673: [Problem] Grayed out/disabled icons do not change when switching style

MisterMakerNL, 2023-05-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32319 (Git)
Build type: Release
Branch: master
Hash: fc996571cb5debcff4643f0eae93a72e632033eb
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Assembly3 0.11.4
  * Behave-Dark-Colors 0.1.1
  * BIM 2021.12.0 (Disabled)
  * Dracula 0.0.4
  * ExtremeProDark 2.7.6
  * ExtremeProLight 1.5.0
  * FeedsAndSpeeds 0.5.0
  * FEM_FrontISTR 0.1.0
  * IconThemes
  * lattice2 1.0.0
  * Nodes 0.1.36
  * PieMenu 1.2.7
  * ProDarkThemePreferencePack 1.0.0
  * Render 2023.2.2
  * scanObjects
  * SelectorToolbar
  * sheetmetal 0.2.59
  * SP_Startup
  * ToolbarStyle
[/code]
```

### Subproject(s) affected?

None

### Problem description

As I am working on updating theme's I was annoyed that the grey out icons where not clear enough.
Some Icons like the new file icon almost have no color, only a tint of yellow.
Also lots of other icons have only 1 extra color. So I felt like the grey wasn't cutting it. I am pretty sure this causes eye strain and annoying to a user they wouldn't even understand why since it is pretty subtle.
Now[ QT documentation]( https://doc.qt.io/qt-5/qicon.html) says they have something in place to account for this issue :

>  The default engine scales pixmaps down if required, but never up, and it uses **the current style to calculate a disabled appearance.** 

But if I compare the icons in 0.21 with dark and light I see this:
[image]
(I have cut them in half and put them side by side.)
So what QT says about the using current style doesn't work or we did something wrong.

### Anything else?

Forum topic:
https://forum.freecad.org/viewtopic.php?t=78431&e=1&view=unread#unread

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2023-05-28:
Did some more reading, not sure if QT can update the disabled icons on the fly.
I do see an option to make the icons transparant which probably also would fix the issue pretty easy.

luzpaz, 2024-08-24:
@MisterMakerNL is this still relevant ?

MisterMakerNL, 2024-08-24:
It is. But since the lack of any reactions here it might not be the most important thing.
Although I am pretty sure it will make the dark style a lot prettier.

luzpaz, 2025-02-01:
Lack of reaction doesn't only mean consensus. We need confirmation that this is not of interest or else UI tickets will be closed by the bot.

MisterMakerNL, 2025-02-01:
This is an issue that nobody notices untill it is fixed.

closed 2025-06-13 by github-actions[bot]