FreeCAD/FreeCAD #9676: Spreadsheet using $ on cell references produces errors on file open

hatkyinc2, 2023-05-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
I have tested this in 0.20.1, 0.20.2, 0.21-weekly 33303-2023-05-25-conda-Windows-x86_64-py310.7z

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33303 (Git)
Build type: Release
Branch: master
Hash: 5b075a9938b78076fdde9034de4a05685e42690e
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * fasteners 0.4.56
[/code]

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * fasteners 0.4.56
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Problem description

Using the `$` sign on a cell, saving the file, and reloading the file produces an error.
Updating the cell with space and saving fixes it.

Recreate:
* New file
* New spreadsheet
* Cell A1 = 1 (anything)
* Cell B1 = `=$A1` (also the same for `=A$1`
* Save the Spreadsheet
* Close file
* Open the file
= Error

Why did I use the $ cell? I got accustomed to it from Excel/google sheets, etc., to move formulas around...

Might be related to https://github.com/FreeCAD/FreeCAD/issues/7645

### Anything else?

[image]
Sample file:
[test.zip](https://github.com/FreeCAD/FreeCAD/files/11574907/test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2023-07-25:
So looked at this. The issue is, that a cell might rely on another cell. If this is the case and it doesn't compute the dependency cell first, then it throws an error. The error is thrown at https://github.com/FreeCAD/FreeCAD/blob/6c72f80d55ebe0bb5549048938277d3eb36f78d3/src/App/ObjectIdentifier.cpp#L1831 under the call
```cpp
Sheet::recomputeCells ->
Sheet::updateProperty ->
Expression::eval ->
Expression::getPyValue ->
VariableExpression::_getPyValue ->
ObjectIdentifier::getPyValue ->
ObjectIdentifier::ResolveResults::ResolveResults ->
ObjectIdentifier::Resolve
```
When there is this many calls just to resolve an expression, then I get a bit worried. Many of them are also just shallow functions for some reason...

The solution to this bug is to implement a guard clause, that checks if the expression has any dependencies and if so computes these first. Unfortunately, I can't really figure out what happens in ObjectIdentifier::Resolve and where we might be able to check for dependencies. And even if it's possible to do a call all the way back to the spreadsheet, as such a guard clause would only be relevant for spreadsheet expressions?

Perhaps @wwmayer could give some guidance?

luzpaz, 2025-01-22:
Re-opening and asking for someone to review if this is actually still relevant

benj5378, 2025-01-24:
Tested, and still an issue.

luzpaz, 2025-01-26:
@mwganson any chance you would be able to assist here? (https://github.com/FreeCAD/FreeCAD/issues/9676#issuecomment-1650042670)

mwganson, 2025-01-26:
When I load the test file there is an initial error message, but upon recomputing the spreadsheet looks okay.  Note that the tree icon clearly shows the need to recompute.

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39896 (Git) Conda
Build type: Release
Branch: main
Hash: a977fade2de741c94c9efa59b1c2c26df18eb631
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.53
  * DynamicData 2.70.0
  * fasteners 0.5.35
  * lattice2 1.0.0
  * MeshRemodel 1.10.36
  * QuickMeasure 2022.10.28
  * ThreadProfile 1.92.0

Is there documentation on the wiki for the use of the $ in spreadsheets?  I did not find it in the wiki, but it is possible I overlooked it.  What is the difference between using A$1 or $A$1 instead of A1 in FreeCAD?

benj5378, 2025-01-28:
> What is the difference between using A$1 or $A$1 instead of A1 in FreeCAD?

@mwganson $ before column means column will stay the same despite the cell contents being moved to another cell. The same goes for $ before rows. Else, moving the cell will auto convert cell references keeping the relative position. https://help.libreoffice.org/24.8/en-US/text/scalc/guide/relativ_absolut_ref.html

=B1 in C1. Moving C1 to D2, content now =C2
=$B1 in C1. Moving C1 to D2, content still =$B2

FEA-eng, 2025-09-14:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/09 06:43:51
Build type: Release
Branch: main
Hash: c1dd116e1e3304511d9e5e634e711e549b2b84e7
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

KBentley57, 2026-08-30:
This bug still effects me on Debian 13, using the 1.1.3 and 1.1.1 app image (Libs: 1.1.1R20260414, Git shallow) and 1.1.3. Opening an FCStd file with a formula like =$B$2 == 0 ? B14 : B14 + C14 gives Property '$B$2' not found in '$B$2' and then says the spreadsheet failed to restore. Changing $B$2 to B2 fixes it. The ternary expression is fine, and the FCStd archive itself is not corrupted.