FreeCAD/FreeCAD #9680: [Problem] Running freecad integrated tests leaves leftover files

Pesc0, 2023-05-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
latest build
```

### Subproject(s) affected?

None

### Problem description

Running `FreeCAD -t 0` leaves behind 
- gcode.tmp:
```
G90
G21
M60

```
- AM_INSTALLATION_DIGEST.txt:
```
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-05-28:
Yes, I also would add:

```ruby
SUMMARY: LeakSanitizer: 50339002 byte(s) leaked in 219478 allocation(s).
```

Pesc0, 2023-05-28:
It seems to me python tests could get a bit of attention and refinement. I wouldn't mind working on this. As part of this would it be a good idea to move said tests outside of freecad, to the tests directory, alongside cpp tests? I did ask this question here: https://github.com/FreeCAD/FreeCAD/pull/9663#issuecomment-1565310332, the idea being:
- All tests would be in the same place, well organized and easy to find.
- Detaching from freecad allows more flexibility, in particular for the goal of better integration with IDEs (#9663) 
- It's one less workbench in freecad. Correct me if I'm wrong, but this is used only by developers, end users don't really need to run tests, right?

Would there be any downsides to this move? Is it quite straight forward or am I committing to a massive endeavor?

adrianinsaval, 2023-05-31:
I don't know, we've gotten several reports that lead to fixing things from users running tests on different platforms (that we don't normally test on).

luzpaz, 2023-09-29:
Other leftover files related ticket: #10221

closed 2025-01-22 by github-actions[bot]