FreeCAD/FreeCAD #9686: No tool in tool controller

CeeDee54, 2023-05-28:
When doing a path operation in 021 release 04-25 & 05-21 under win 10, latest version & updates, it doesn't give the tool list. Tool controller is present but non functional. Whether opening an existing path operation or opening a new operation.  (It worked a week or so ago.) 

It gives this error.
```
06:55:23  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD_weekly-builds-33303-2023-05-25-conda-Windows-x86_64-py310\Mod\Path\Path\Op\Gui\Base.py", line 107, in setEdit
    self.setupTaskPanel(
  File "C:\Program Files\FreeCAD_weekly-builds-33303-2023-05-25-conda-Windows-x86_64-py310\Mod\Path\Path\Op\Gui\Base.py", line 120, in setupTaskPanel
    panel.setupUi()
  File "C:\Program Files\FreeCAD_weekly-builds-33303-2023-05-25-conda-Windows-x86_64-py310\Mod\Path\Path\Op\Gui\Base.py", line 1307, in setupUi
    self.panelSetFields()
  File "C:\Program Files\FreeCAD_weekly-builds-33303-2023-05-25-conda-Windows-x86_64-py310\Mod\Path\Path\Op\Gui\Base.py", line 1262, in panelSetFields
    page.pageSetFields()
  File "C:\Program Files\FreeCAD_weekly-builds-33303-2023-05-25-conda-Windows-x86_64-py310\Mod\Path\Path\Op\Gui\Base.py", line 249, in pageSetFields
    self.setFields(self.obj)
  File "C:\Program Files\FreeCAD_weekly-builds-33303-2023-05-25-conda-Windows-x86_64-py310\Mod\Path\Path\Op\Gui\Profile.py", line 95, in setFields
    self.setupToolController(obj, self.form.toolController)
  File "C:\Program Files\FreeCAD_weekly-builds-33303-2023-05-25-conda-Windows-x86_64-py310\Mod\Path\Path\Op\Gui\Base.py", line 390, in setupToolController
    controllers = PathUtils.getToolControllers(self.obj)
  File "C:\Program Files\FreeCAD_weekly-builds-33303-2023-05-25-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathUtils.py", line 367, in getToolControllers
    return [tc for tc in job.Tools.Group if proxy.isToolSupported(obj, tc.Tool)]
  File "C:\Program Files\FreeCAD_weekly-builds-33303-2023-05-25-conda-Windows-x86_64-py310\Mod\Path\PathScripts\PathUtils.py", line 367, in <listcomp>
    return [tc for tc in job.Tools.Group if proxy.isToolSupported(obj, tc.Tool)]
```

luzpaz, 2023-05-28:
Transferred from FreeCAD-bundle to FreeCAD core issue queue to make sure this is only AppImage related.

CeeDee54, 2023-05-28:
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33321 (Git)
Build type: Release
Branch: master
Hash: 1d61ed187c9633702818c6cc5d7a1f07dd052ca8
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.59e
  * Curves 0.5.13
  * fasteners 0.4.16
  * freecad.gears 1.0.0
[/code]

I downloaded the 2023-05-028 win py310 & py38.
Doesn't work on my Win 11 laptop either.

Syres916, 2023-05-29:
I cannot replicate this bug using:

```
OS: Windows 7 build 7601
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.8.16, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.60f
  * Alternate_OpenSCAD 1.0.0
  * Assembly3 0.11.4
  * Assembly4 0.12.6
  * CfdOF 1.20.4
  * Curves 0.6.6
  * fasteners 0.4.54
  * fcgear 1.0.0
  * freecad.xray 2022.4.17
  * FreeCAD_Turning_Addon 0.1.0
  * Help 1.0.3
  * Manipulator 1.5.0
  * PitchedRoof
  * Plot 2022.4.17
  * Render 2023.2.2
  * sheetmetal 0.2.63
  * Silk 1.0.0
  * timber
  * WorkFeature

```

If it was a widespread issue I would have expected the Path subforum to be well aware by now and I cannot see any evidence of that but willing to be corrected.

CeeDee54, 2023-05-29:
Now I've tried a 3rd file & the tool worked, though its a simple process. So it's obvious to me that the first 2 files are corrupted. The question is how did they get this way? What can I do to restore them? Could it be with relation to the add tools from the docked tool list that doesn't seam to have a good flow? 

I can't seam to attach the primary working FCStd file relevant to this problem. The zip file is too large, 61MB. I have attached a modified version for the text carving that also has the tool problem.
[GP sfcstd1.zip](https://github.com/FreeCAD/FreeCAD/files/11593935/GP.sfcstd1.zip)
[image]
[GP fcstd.zip](https://github.com/FreeCAD/FreeCAD/files/11593934/GP.fcstd.zip)

sliptonic, 2023-05-30:
The workflow with regard to Dock is being revisited.   I'm not sure what caused the problem but repairing it is pretty easy
https://github.com/FreeCAD/FreeCAD/issues/9466

[image]

Closing this since we don't seem to have anything actionable right now.

closed 2023-05-30 by sliptonic

CeeDee54, 2023-05-30:
thank you. I've now got them both working. There's always something to revisit during development. All the best.