FreeCAD/FreeCAD #9692: Crash on macOS when filesystem is case sensitive (icudtl.dat and qtwebengine* nor found)

AKielhorn, 2023-05-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 11.7.7
Word size of FreeCAD: 64-bit
Version: 0.21.0.33321 (Git)
Build type: Release
Branch: master
Hash: 1d61ed187c9633702818c6cc5d7a1f07dd052ca8
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * Assembly4 0.12.6
  * Plot 2022.4.17
  * BOLTSFC 2022.11.5
  * MnesarcoUtils 0.2.1
  * A2plus 0.4.60k
  * fcgear
  * FreeCAD_assembly3 0.11.3
  * fasteners 0.4.55
  * lattice2 1.0.0
  * sheetmetal 0.2.61
[/code]
```

### Subproject(s) affected?

None

### Problem description

FreeCAD on macOS crashes as described here:

[MacOS at startup crash (critical)](https://forum.freecad.org/viewtopic.php?t=71835)

The error message is:

```
Qt WebEngine resources not found at /Applications/FreeCAD.app/Contents/Resources/Resources. Trying parent directory...
```

When I look into the application I see

```
/Applications/FreeCAD.app/Contents/Resources/resources
```

with a lowercase "r".

If I rename the folder to "Resources" it works.

This affects only a small number of macOS users, since the default on macOS is to ignore case.

Greetings
Axel

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-05-29:
Is this the Conda macOS build? (where did you install from?)

AKielhorn, 2023-05-29:
Yes, it is the conda build.

Installed from
[weekly-builds](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

luzpaz, 2023-05-30:
Is this the relevant code? 
https://github.com/FreeCAD/FreeCAD-Bundle/blob/master/conda/osx/create_bundle.sh#L54-L55

for arm64:  
https://github.com/FreeCAD/FreeCAD-Bundle/blob/master/conda/osx-arm64/create_bundle.sh#L57-L58

AKielhorn, 2023-05-30:
Yes that seems to fix it.

When I
```
ln -s resources Resources
```

I get

```
Resources % ls -ld [rR]*
lrwxr-xr-x  1 ***  admin    9 30 Mai 18:14 Resources@ -> resources
drwxr-xr-x@ 8 ***  admin  256 28 Mai 14:30 resources/
```

and FreeCAD starts just fine.