FreeCAD/FreeCAD #9698: Windows build throws `QOpenGLShader::compile(Fragment): ERROR: 4:1: 'textureSampler' : syntax error syntax error` (Upstream Qt issue)

snelweg, 2023-05-30:
This error is back. The Windows version from the installer package works fine, the zipped version gives this error.

```
Wrn: QOpenGLShader::compile(Fragment): ERROR: 4:1: 'textureSampler' : syntax error syntax error

Wrn: *** Problematic Fragment shader source code ***
#version 110
#ifdef GL_KHR_blend_equation_advanced
#extension GL_ARB_fragment_coord_conventions : enable
#extension GL_KHR_blend_equation_advanced : enable
#endif
#define lowp
#define mediump
#define highp
#line 1
varying highp vec2 uv;uniform sampler2DRect textureSampler;uniform bool swizzle;uniform highp float opacity;void main() {   highp vec4 tmpFragColor = texture2DRect(textureSampler,uv);   tmpFragColor.a *= opacity;   gl_FragColor = swizzle ? tmpFragColor.bgra : tmpFragColor;}
***
Wrn: Could not link shader program:
 "ERROR: 4:1: 'textureSampler' : syntax error syntax error\n\n"
Log: Loading Spreadsheet module... done
Log: Loading GUI of Spreadsheet module... done
```

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32662 (Git)
Build type: Release
Branch: master
Hash: 312fe9b5c822126b251ca1c3cfb55dde0de1920c
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods: 
  * Assembly4 0.12.6

```

[FreeCAD.log](https://github.com/FreeCAD/FreeCAD-Bundle/files/11135429/FreeCAD.log)

luzpaz, 2023-05-30:
Transferred from FreeCAD-Bundle to FreeCAD main issue queue for exposure since this is upstream Qt related

JulienMaille, 2023-05-31:
> @JulienMaille thanks for the heads-up!

I opened an issue and a merge request over there: https://invent.kde.org/qt/backports-tracker/-/issues/2446

closed 2025-01-19 by github-actions[bot]