FreeCAD/FreeCAD #9701: [Problem]

berniev, 2023-05-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
na
```

### Subproject(s) affected?

None

### Problem description

Further to #8657 the current structure of PR introduction is being ignored or deleted in many cases, including by senior members of this project.

The checkmark (place an 'x' to make a tick [huh?]) is counterintuitive and doesn't work if a leading or trailing space. An interesting UI design failure.

And who's going to read all that text, except maybe a first-timer. Maybe.

With no reward for jumping through these hoops and no punishment for not jumping through them, it is only natural they will be ignored, and possibly resented as cognitive distractions.

IMHO the whole text should be reduced to a couple of succinct, polite, lines and a couple of links.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-05-30:
Please add a legit title to this ticket l, TIA

berniev, 2023-05-30:
Oops! Tks!

adrianinsaval, 2023-05-31:
why create a separate issue for this and not jsut add this as comment in that issue?

berniev, 2023-05-31:
Because this report is addresses the **fact** that lots of PRs are ignoring the introduction, and attempting to identify the cause and a possible solution, not just expressing a thought.

benj5378, 2023-07-23:
This should be a forum discussion and not an issue on GitHub. @berniev, please start a new thread on the forum with proposal of a new text.

berniev, 2023-08-10:
No, the policy of flicking issues back to forum was abandoned months ago.

luzpaz, 2023-08-12:
I'm tryting to convert this issue to a discussion and there is a problem. We need to do it manually.

closed 2023-08-12 by luzpaz