FreeCAD/FreeCAD #9707: [Problem] FEM Example Capacitance Two Balls malfunction in FreeCAD v 0.20.2

Ziemnianin007, 2023-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

FEM

### Problem description

I tried to use template [FEM Example Capacitance Two Balls](https://wiki.freecad.org/FEM_Example_Capacitance_Two_Balls) and reproduce the results in newest stable FreeCAD release 0.20.2 (on windows 11). I made clean installation of software to be sure that no other settings or addons influence results but it still comes the same. It seems that the results are somehow reversed:
[image]
To check if the steps from tutorial were performed fine i repeated process in older stable version of freecad 0.19.3 (downloaded from this [thread](https://forum.freecad.org/viewtopic.php?style=4&p=571167) as the release versions from github cannot open examples, and this is the newest working) those are correct results:
[image]

Especially interesting for me is the capacitance matrix results which for newer version are wrong:
FreeCAD V 0.20.2
StatElecSolver: Capacitance matrix computation performed (i,j,C_ij)
StatElecSolver:   1  1   -5.36170E-11
StatElecSolver:   1  2    6.48460E-11
StatElecSolver:   2  2   -5.35964E-11
And those are correct:
FreeCAD V 0.19.3 
StatElecSolver: Capacitance matrix computation performed (i,j,C_ij)
StatElecSolver:   1  1    5.07048E-06
StatElecSolver:   1  2    1.69565E-06
StatElecSolver:   2  2    5.07580E-06

It seems that with upgrading from v 0.19 to v 0.20 something in EM FEM simulation has been broken.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-06-22:
I can confirm that the issue occurs in the newest weekly build of FreeCAD 0.21 as well (details below). It's due to the fact that all 3 electrostatic potential boundary conditions don't have the "unspecified" box next to the "Potential" field checked in the newer versions.

[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]

marioalexis84, 2024-01-21:
As @FEA-eng stated, the property `PotentialEnabled` property need to be setted to `False` for the electrostatic potential boundary condition (in fact, is necessary and sufficient to do that for the large sphere, to properly set the boundary condition at infinity). This is solved with 

Two things to note:
* The FreeCAD Elmer writer currently use SI units, so the vacuum permittivity is setted to $\epsilon_{0} = 8.85  \text{ x } 10^{-12} \frac{F}{m}$.
In this units system, the simulation returns the following values for the capacitance coefficients (given in Farad):
  ```
  StatElecSolver:   1  1    4.46299E-11
  StatElecSolver:   1  2    1.48335E-11
  StatElecSolver:   2  2    4.47960E-11
  ```
  Which are very nearly to the values given by the approximate analytical solution (see [this reference](https://www.iue.tuwien.ac.at/phd/wasshuber/node77.html)): 
  $C_{12} \simeq 1.497 \text{ x } 10^{-11} F$
  $C_{10} \simeq 4.444 \text{ x } 10^{-11} F$

  The values reported in the [documentation](https://wiki.freecad.org/FEM_Example_Capacitance_Two_Balls) are obtained using a vacuum permittivity $\epsilon_0 = 1$, identically to the used in the [Elmer tutorial](https://www.nic.funet.fi/pub/sci/physics/elmer/doc/ElmerTutorials.pdf#chapter.12).

* In the second part of the example, the result is post processing  using a region clip filter, but the result obtained have nothing to do with the potencial values setted in the electrostatic potential boundary condition: If the property `CalculateCapacitanceMatrix` is `True`, The potencial values are ignored by Elmer and configured internally to calculate the capacitance coefficients (which depend only on the geometry and not the charge/potential state of the system).
The last step of Elmer's iteration set the potential to one for the las object (`Cut.Face3`) and to zero the other (in this case `Cut.Face2`)). So in the example, the fields values showed by the filter and the pipeline are the used by the latest iteration of the solver.
This is very confussing: One thing that the user could do to play with this example is change the potential values to obtain new fields, but since the capacitance matrix is being calculated, systematically the result will be the latter: potential equal to one on `Face3` and zero on `Face2`.
Because of this, the example should be split in two steps: First calculate the capacitance coefficients using `PotentialEnabled = False` and in the second step set `CalculateCapacitanceMatrix = False` and play with potential values of the boundary condition and from there generate the post processing filters.

commit 629f585a (2024-01-21): Fem: Set PotentialEnabled to False in capacitance_two_balls example - fixes #9707
closed 2024-01-21 by chennes