FreeCAD/FreeCAD #9708: Creating a pocket with a negative taper crashes freecad when the ketch is a circle with radius 1mm

electroCutie, 2023-06-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
freecad/kinetic,kinetic,now 0.20.1+dfsg1-2 all
```

### Subproject(s) affected?

PartDesign

### Problem description

When making a pocket from a circle with a radius of exactly 1 mm and having any negative angle freecad crashes

I've recreated the issue with a fresh file and create a minimum example
• Create a cube
• Place a sketch on a face (I chose the front one but I do not believe that matters)
• Create a pocket with a negative angle (one that is feasible and doesn't come to a point, not even close)
That is the setup, now the bug
• Trigger the error by setting a restriction on the circle with a radius of exactly 1mm

Hard crash (instantly closed window, sigsev seen on the output)

Here is the minimal example described above. Edit the sketch within the pocket and set the radius to exactly 1mm to trigger the crash, however this crash seems to trigger quite reliably
[cursedCube.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11627728/cursedCube.FCStd.zip)

### Anything else?

The most similar issue I found was #9238 but offhand they seem to have different triggers and share only passing similarity as my bug involves a pocket and theirs an extrusion but perhaps there is an internal commonality there. I am not qualified to know.

Here is the full CLI output when triggering the error.

FreeCAD 0.20.1, Libs: 0.20.1R
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Interlocking laser cut workbench loaded
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bcf0) [0x7fb4ec43bcf0]
#1  0x7fb43d6ad30c in MAT_ListOfBisector::Current() const from /lib/x86_64-linux-gnu/libTKTopAlgo.so.7+0xc
#2  0x7fb43d6c4635 in MAT2d_Mat2d::CreateMat(MAT2d_Tool2d&) from /lib/x86_64-linux-gnu/libTKTopAlgo.so.7+0x2df5
#3  0x7fb43d6e542d in BRepMAT2d_BisectingLocus::Compute(BRepMAT2d_Explorer&, int, MAT_Side, GeomAbs_JoinType, bool) from /lib/x86_64-linux-gnu/libTKTopAlgo.so.7+0xbad
#4  0x7fb43dc1b1b0 in BRepFill_OffsetWire::Init(TopoDS_Face const&, GeomAbs_JoinType, bool) from /lib/x86_64-linux-gnu/libTKBool.so.7+0x190
#5  0x7fb43dc1b5db in BRepFill_OffsetWire::BRepFill_OffsetWire(TopoDS_Face const&, GeomAbs_JoinType, bool) from /lib/x86_64-linux-gnu/libTKBool.so.7+0xfb
#6  /lib/x86_64-linux-gnu/libTKOffset.so.7(+0x126e13) [0x7fb4867cce13]
#7  0x7fb486713d7e in BRepOffsetAPI_MakeOffset::Perform(double, double) from /lib/x86_64-linux-gnu/libTKOffset.so.7+0x48e
#8  0x7fb43cd2062f in Part::ExtrusionHelper::createTaperedPrismOffset(TopoDS_Wire, gp_Vec const&, double, bool, TopoDS_Wire&) from /usr/lib/freecad-python3/lib/Part.so+0x24f
#9  0x7fb43cd220cf in Part::ExtrusionHelper::makeDraft(TopoDS_Shape const&, gp_Dir const&, double, double, double, double, bool, std::__cxx11::list<TopoDS_Shape, std::allocator<TopoDS_Shape> >&, bool) from /usr/lib/freecad-python3/lib/Part.so+0x179f
#10  0x7fb488eb1be8 in PartDesign::FeatureExtrude::generateTaperedPrism(TopoDS_Shape&, TopoDS_Shape const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, gp_Dir const&, double, double, double, double, bool) from /usr/lib/freecad-python3/lib/_PartDesign.so+0x2e8
#11  0x7fb488ec3474 in PartDesign::Pocket::execute() from /usr/lib/freecad-python3/lib/_PartDesign.so+0x994
#12  0x7fb4ee99b928 in App::DocumentObject::recompute() from /usr/lib/freecad-python3/lib/libFreeCADApp.so+0x3a8
#13  0x7fb43cbe881f in Part::Feature::recompute() from /usr/lib/freecad-python3/lib/Part.so+0xf
#14  0x7fb4ee9648bc in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad-python3/lib/libFreeCADApp.so+0x22c
#15  0x7fb4ee968e10 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib/freecad-python3/lib/libFreeCADApp.so+0x570
[... 75 lines cut ...]
#91  0x7fb4ecebae38 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#92  0x7fb4edb726b2 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1c2
#93  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c80dc) [0x7fb4edbc80dc]
#94  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cb4a0) [0x7fb4edbcb4a0]
#95  0x7fb4edb6bf32 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x82
#96  0x7fb4ef20d050 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x90
#97  0x7fb4ecebae38 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#98  0x7fb4ed3426fd in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x80d
#99  0x7fb4ed31663c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#100  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x7478e) [0x7fb4e6d6378e]
#101  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x299) [0x7fb4eaf21569]
#102  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab3c8) [0x7fb4eaf763c8]
#103  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x30) [0x7fb4eaf1ed20]
#104  0x7fb4ecf14aea in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x6a
#105  0x7fb4eceb97cb in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#106  0x7fb4ecec1c2a in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x9a
#107  0x7fb4ef1a639a in Gui::Application::runApplication() from /usr/lib/freecad-python3/lib/libFreeCADGui.so+0x1eba
#108  freecad(+0x50e5) [0x55f7f13ff0e5]
#109  /lib/x86_64-linux-gnu/libc.so.6(+0x23510) [0x7fb4ec423510]
#110  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x89) [0x7fb4ec4235c9]
#111  freecad(+0x60f5) [0x55f7f14000f5]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-01:
Please post actual full version information as provided by the application in Help->About->Copy to clipboard, the details there are important.

Anyways, please try the appimage, flatpak or snap releases as those use a patched version of occt that does not crash as often, plus the version in the package repo is likely outdated too

electroCutie, 2023-06-01:
Updated the version information and also switched to the snap version 

The snap version doesn't have this bug. Thank you for the suggestion, that keeps me going

[code]
OS: Ubuntu Core 20 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git) Snap 517
Build type: Release
Branch: tag: 0.20.2
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.5, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
[/code]

electroCutie, 2023-06-01:
Ah, the snap version has a related bug! It doesn't crash but it refuses to perform the action

Same steps to reproduce, same file even, but now we get an error in the log:

```
23:44:54  Extrusion: end face of tapered along extrusion is empty
This means most probably that the along taper angle is too large or small.
```

This is not true, there is still plenty of bottom left in there. Making the circle larger, making the cut shallower, making the negative taper 1°, etc all lead to the same error

I can also say that this happens at or below 1mm, not just at 1mm exactly. So this still seems like a bug unless I'm missing something. it doesn't crash in the snap version but its still wrong

adrianinsaval, 2023-06-02:
I think there is a general bug in how circles are offset by occt, IIRC there where previous reports of failure or inconsistent tapereded extrusions with circles

krushia, 2024-07-25:
I did a check on this; still broken as described.

Whether this crashes or shows an error seems to depend on how FreeCAD was built. I'm getting the error on the weekly AppImage[1] and the stable (0.21.3) Flatpak[2], but the crash on self-built source[3].

[1]
```
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git) AppImage
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

[2]
```
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.7, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
```

[3]
```
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.Unknown (whatever was in tree yesterday)
Build type: RelWithDebInfo
Python 3.11.9, Qt 5.15.14, Coin 4.0.2, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

arcivanov, 2024-10-08:
I have the same issue with a Pad negative taper crashing 0.21.2, latest dev and RealThunder

Beep6581, 2024-11-01:
Reproduced on:
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39046 (Git)
Build type: Release
Branch: main
Hash: b6cda4309a9031bebffa73b694e9a9cc125d64de
Python 3.12.6, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * OpenDark 2023.12.17
  * fasteners 0.5.25
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.9
  * Curves 0.6.42

```
0x00007fff8f8c56db in MAT_ListOfBisector::Current() const () from /usr/lib/libTKTopAlgo.so.7.8
#1  0x00007fff8f8e2b9a in MAT2d_Mat2d::CreateMat(MAT2d_Tool2d&) () from /usr/lib/libTKTopAlgo.so.7.8
#2  0x00007fff8f8ff740 in BRepMAT2d_BisectingLocus::Compute(BRepMAT2d_Explorer&, int, MAT_Side, GeomAbs_JoinType, bool) () from /usr/lib/libTKTopAlgo.so.7.8
#3  0x00007fff98a51ee7 in BRepFill_OffsetWire::Init(TopoDS_Face const&, GeomAbs_JoinType, bool) () from /usr/lib/libTKBool.so.7.8
#4  0x00007fff98a523fd in BRepFill_OffsetWire::BRepFill_OffsetWire(TopoDS_Face const&, GeomAbs_JoinType, bool) () from /usr/lib/libTKBool.so.7.8
#5  0x00007fff9a1cadb4 in ?? () from /usr/lib/libTKOffset.so.7.8
#6  0x00007fff9a0f93f3 in BRepOffsetAPI_MakeOffset::Perform(double, double) () from /usr/lib/libTKOffset.so.7.8
#7  0x00007fff9a6d39cb in Part::ExtrusionHelper::createTaperedPrismOffset(TopoDS_Wire, gp_Vec const&, double, bool, TopoDS_Wire&) () from /home/morgan/programs/freecad/lib/Part.so
#8  0x00007fff9a6d61e2 in Part::ExtrusionHelper::makeElementDraft(Part::ExtrusionParameters const&, Part::TopoShape const&, std::vector<Part::TopoShape, std::allocator<Part::TopoShape> >&, Base::Reference<App::StringHasher>) () from /home/morgan/programs/freecad/lib/Part.so
#9  0x00007fff8de1c7e6 in PartDesign::FeatureExtrude::buildExtrusion(Base::Flags<PartDesign::FeatureExtrude::ExtrudeOption>) () from /home/morgan/programs/freecad/lib/_PartDesign.so
#10 0x00007ffff4903c6a in App::DocumentObject::recompute() () from /home/morgan/programs/freecad/lib/libFreeCADApp.so
```

TechnischeEntwicklungenGretsch, 2026-01-13:
this Issue seems to be related to #22391

stefano2734, 2026-09-12:
It is Solved in OCCT 8.0.2, planned for October 2026.
https://github.com/Open-Cascade-SAS/OCCT/milestone/8?closed=1

ElementW, 2026-09-13:
Fixed as of OCC 8.1.0.dev1-3d097a0328 (FC 13b07d70e6e27ee61f8894531c22aeeb46aa93ed).

@stefano2734 You could've pointed to the actual PR that solved it: https://github.com/Open-Cascade-SAS/OCCT/pull/1453

FWIW #22391 is a duplicate of this, stack trace is basically the same and the setup is just in boolean union instead of boolean difference.

luzpaz, 2026-09-16:
cc @pieterhijma 👆

pieterhijma, 2026-09-30:
Marked as Done in the OCCT Liaison board.

closed 2026-10-27 by pieterhijma

luzpaz, 2026-10-27:
AKA Fixed in upstream OCC v8.0.2