FreeCAD/FreeCAD #9713: Problema de escritura

NIT-PHOS, 2023-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Peru (es_PE)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Problema de escritura, al crear un cubo, si hacemos click en el  en las opciones de abajo, vemos la pestaña de datos, en la sección MAP MODE dice "Deactivated"

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

YosypC, 2023-06-02:
Please type in English, we can't help you otherwise!

adrianinsaval, 2023-06-06:
> Problema de escritura, al crear un cubo, si hacemos click en el en las opciones de abajo, vemos la pestaña de datos, en la sección MAP MODE dice "Deactivated"

fotos? que se supone que es el problema? Y por favor responde en ingles, usa el traductor de google si hace falta que no somos muchos los hispanohablantes desarrollando FreeCAD.

pictures? what is supposed to be the problem?

luzpaz, 2023-06-07:
Closing ticket. If @NIT-PHOS can provide more info, they can re-open.

closed 2023-06-07 by luzpaz