FreeCAD/FreeCAD #9714: [Problem] Project Utility tool does not work well at all

YosypC, 2023-06-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
```

### Subproject(s) affected?

None

### Problem description

I am experimenting with Git and FC without Large File Support, so I resorted in unpacking the projects with the Project Utility tool.
The unpacking seems to work, but the repacking in FCStd has some serious issues:

1. The resulted project is ALWAYS called "project", regardless of its original name.
2. The resulted project has .fcstd as a file extension instead of .FCStd .
3. You can't unpack a file containing an .FCStd dependency, as it gives an error despite the .FCStd present in destination folder:
[image]
4. You can't automate multiple actions without Pythoning.
5. You can't delete a failed project.fcstd repack despite having every document closed in FC, resorting in FC closure.
[image]

I've added my working folders too so you can give it a look yourself.

### Anything else?

[DependentFiles.zip](https://github.com/FreeCAD/FreeCAD/files/11638367/DependentFiles.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

YosypC, 2023-06-08:
@luzpaz [Done!](https://forum.freecad.org/viewtopic.php?t=78892)

Zolko-123, 2023-06-08:
What is "_Project Utility tool_ " ?

YosypC, 2023-06-08:
@Zolko-123 [This!](https://wiki.freecad.org/Std_ProjectUtil) I invite you to discuss it on the forum for a more open discussion

Zolko-123, 2023-06-08:
Then I don't understand: the documentation clearly states that : 

> A new project file with a fixed name, project.fcstd, is created in the selected folder. 

and yet you complain exactly about that:

> 1. The resulted project is ALWAYS called "project", regardless of its original name.
> 2. The resulted project has .fcstd as a file extension instead of .FCStd

So it seems to work as advertised

YosypC, 2023-06-08:
> So it seems to work as advertised

"It's not a bug, it's a feature" is what I am reading. It should be improved because the way it works now is not enough even for basic documents

adrianinsaval, 2023-06-08:
hardcoded project.fcstd:
https://github.com/FreeCAD/FreeCAD/blob/16b06993e23eabd411586c7908be08ad8f609ce0/src/Gui/DlgProjectUtility.cpp#L90

I guess we could prompt the user for a filename instead

adrianinsaval, 2023-06-08:
> You can't unpack a file containing an .FCStd dependency, as it gives an error despite the .FCStd present in destination folder:

Can you explain better how to reproduce this? Step by step instructions please

> You can't automate multiple actions without Pythoning.

what do you mean? how else are you expecting to automate?

> You can't delete a failed project.fcstd repack despite having every document closed in FC, resorting in FC closure.

please provide step by step details on how to reproduce

luzpaz, 2024-07-12:
@YosypC :point_up:

luzpaz, 2024-07-22:
@YosypC once all the [1.0 blocker bugs](https://github.com/FreeCAD/FreeCAD/issues?q=is%3Aopen+is%3Aissue+milestone%3A1.0+label%3ABlocker) are fixed 1.0 will be released

luzpaz, 2025-06-13:
Should be addressed

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/1c9babeecf55650d8bc9f174230fc890e210afdb

Extraction accepts existing subdirectories and archives are closed after a failure. The hardcoded project filename and batch automation are not changed.