FreeCAD/FreeCAD #9716: [Problem] Mesh Workbench broken in MacOs

martin1951, 2023-06-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 12.6.6
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * DynamicData 2.46.0
  * Help 1.0.3
  * Curves 0.6.9
[/code]
```

### Subproject(s) affected?

None

### Problem description

The Mesh workbench in the current development version of FreeCAD for MacOS seems to be broken. I'm using a MacBook Air, 2.2 GHz Dual-Core Intel i7 running MacOS version 12.6.6. When attempting to make a mesh from a Part Design model  the Mesh workbench loads, selecting the Make Mesh icon brings up the dialogue box but clicking 'OK' dismisses the dialogue but doesn't produce a mesh. Also noted that Mefisto and Netgen options aren't available in Mesh dialogue. Everything works as expected with build 31641.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-06:
Likely a duplicate of https://github.com/FreeCAD/FreeCAD-Bundle/issues/173

luzpaz, 2023-06-07:
Closing. Please re-open if above related ticket is another issue altogether.

closed 2023-06-07 by luzpaz

martin1951, 2023-06-26:
Just checked current weekly build and still unable to created mesh.
[code]
OS: macOS 12.6.6
Word size of FreeCAD:  @64-bit
Version: 0.21.0.33445 (Git)
Build type: Release
Branch: master
Hash: a3d445e1c9d3706f8ad99fdf8f5902bae1bf3373
Python 3.11.4, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.87.0
  * DynamicData 2.46.0
  * Help 1.0.3
  * Curves 0.6.12
[/code]

adrianinsaval, 2023-06-26:
I'm working on getting conda builds using occt 7.6.3 working again, once that is done I'll change the bundles to use those builds and this should get fixed. We'll still have to deal with this later on conda-forge though, eventually we will want to move to occt 7.7

martin1951, 2023-06-28:
No pressure. Wasn’t sure if this was still on the radar or something unique about my system. Thanks for getting back to me. Kudos for all you do.