FreeCAD/FreeCAD #9725: [Problem] Dragging a Sketch under another body doesn't yield the same result as "Move object to another body"

YosypC, 2023-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
```

### Subproject(s) affected?

None

### Problem description

I initially created a sketch under a body, then realized I needed it somewhere else: usually dragging features in and out of bodies gives mixed results, such as this case where I get "_Bad object "NameOfTheSketch" (Sketcher::SketchObject) linked to the Origin of "NameOfTheDocument#BodyWhereIAmTryingToMoveItTo"_ ".

https://github.com/FreeCAD/FreeCAD/assets/50027752/c8522e0b-1612-4d86-a480-ddf7ced5bcd7

### Anything else?

File: 
[ExpCard_MoboIOPlate.zip](https://github.com/FreeCAD/FreeCAD/files/11645372/ExpCard_MoboIOPlate.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-06-04:
Drag n' drop a plain unattached Sketch is something different as move an attached Sketch from PartDesign to one Body to another. To not break things too easy, that is intended. So no issue.

YosypC, 2023-06-08:
> Drag n' drop a plain unattached Sketch is something different as move an attached Sketch from PartDesign to one Body to another. To not break things too easy, that is intended. So no issue.

Users cannot differentiate the two mechanism the way they are designed now. The wiki pages [Std TreeDrag](https://wiki.freecad.org/Std_TreeDrag) and [PartDesign MoveFeature](https://wiki.freecad.org/PartDesign_MoveFeature) are not explanatory.

grubertechnic, 2023-06-08:
Only because it is not documented, it does not mean that is does not work correct. As i wrote, _Move object to other body_ does much more as only drag an object to one plece to an other. It changes its [attachment](https://wiki.freecad.org/Part_EditAttachment ) (to the target Body origin).

Maybe the difference is not obvious in the first hour, but if you <ins>use</ins> FreeCAD and PartDesign, you will figure it out in an appropriate time.

grubertechnic, 2023-06-08:
Generally, i would close the ticket since it works as expected.

grubertechnic, 2023-06-09:
> Your opinion is understandably biased since you are a developer (if the "Contributor" tag does not lie)

It lies, i am not a developer, i am a user. The little bit i have done here is absolute negligible (max. 30 lines so far, but i am working that it will be more).

> There's no sign, no clue, apart the fact that one action works and the other doesn't, that they are fundamentally two different commands.

So far every program (or CAD) i have worked, dragging or a move command behaves pretty always different, else there is no need for a move command. 

> It needs to be changed or somehow explained to the user first (or then) because the way it works now is only confusing to say the least.

That the documentation and a few hints must be updated is in fact true, nothing against it.

luzpaz, 2023-06-10:
We need more feedback from other FC devs here

YosypC, 2023-06-11:
I've created a [thread](https://forum.freecad.org/viewtopic.php?t=78932) on the forum.

PaddleStroke, 2023-06-12:
Copy pasting and moving actions in the tree are for me sub-standard in their complexity.  As a user it annoyed me a lot. Copy pasting a simple sketch is weirdly difficult.

I don't see why we have a 'move object to other body' in the first place. Drag and drop should be the way it's done.
Same for 'Move object after other object', I think this menu entry should not exist, it should be done by drag & drop.

grubertechnic, 2023-06-12:
> Copy pasting and moving actions in the tree are too complex and not intuitive IMO. As a user it annoyed me a lot. Copy pasting a simple sketch is weirdly difficult.
> 
> I don't see why we have a 'move object to other body' in the first place. Drag and drop should be the way it's done.

Sorry, i know you develop many, but did you use FreeCAD or any other CAD? It had serious reason, why this behave like this (and for me it is intuitive, also till now no one complain about that). FreeCAD is no exception in this. You must handle the dependencies, and "just" moving and copying is in a parametric modeler the exception and nearly not existent.

PaddleStroke, 2023-06-12:
Yes I come from modelling and PCB design. I learned to develop to make some things for FreeCAD. Then it snowballed and the situation is a bit reversed.
Every user has different needs and uses I guess. I fail to see what dependencies you need when trying to copy paste a sketch.

I mean just look at that, here I try to copy sketch003. What a new user is supposed to feel if not overwhelmed : 
[image]
And just try to select only the sketch. How daunting it gets just to select only the sketch.
And this is a body that is not so complex with only 6 operations.

grubertechnic, 2023-06-12:
That the view (window) is a little bit odd and can be optimized is indeed true, but it is needed.

For example you want to copy the sketch unattached on the plane `XY-plane` or complete without dependencies. (Done both 1/2h ago). FreeCAD can not predict what you want and how to copy, as all other CADs can not do. All parametric CADs have a similar dialog while copying or when pasting (which is the same only delayed).

> And this is a body that is not so complex with only 6 operations.

When you resolve the 6 operations, then it is already complicated. The PartDesign conception makes is look simple, there is many done under the hood.

adrianinsaval, 2023-06-12:
There are certainly situations where moving or copying things isn't straightforward but if it's possible to trigger the same algorithm from the move commands but whit a drag action I think that is desirable.

The problem description is genuine, to an user it is not clear why one works and the other doesn't or why they should use one over the other. "Other CADs do it" is not a valid counterargument as a big portion of users do not have prior CAD experience and FreeCAD cannot require previous experience in other software for it's use. The software should strive to be usable on it's own (or with it's own documentation at most).

grubertechnic, 2023-06-17:
> "Other CADs do it" is not a valid counterargument

It is not limited to CAD as i wrote before, even table calculation programs it is similar. There is very rare only a copy option per default (and if then, they are not good). FreeCAD must handle the dependencies and can not guess, also automagic is pretty dangerous in parametric. Have seen too much flaws in real life because of false "guessing" or not handling them, that cost literally millions of $/€ because of this. Better take the 5 seconds and get good and expectable results as repair the whole time flaws. And if you do not know there is a flaw in then dependencies because of automagic, then this i bad situation anyway.

Just said, the construction in CAD itself is still a part of engineering, which cost not that much time. What consumes a lot more time is conception, dimensioning, calculating, search in tables/tablebooks, search in datasheets, interchange construction team and/or sections of other parts/assemblies, interchange with machinist, ...... The CAD only part takes around 25% to 33.33%, just engineering as said. Better this works "slow" and well expectable as fast and furious, but somehow random.

grubertechnic, 2023-06-17:
Just a note, i am not against for any changing, but it must really make sense. The copy dialog as example, again. Not the first change attempt.

adrianinsaval, 2023-06-17:
We're not talking of copy operations. The dependency dialog when copying is necessary. We're talking of performing the same operation when dragging as when using "move to another bofy" the mechanism already exists, it's only a matter of making it happen with a different action.

maxwxyz, 2024-02-25:
Is this a duplicate to https://github.com/FreeCAD/FreeCAD/issues/12055 ?

YosypC, 2024-02-29:
Looks like the same issue.

maxwxyz, 2024-04-18:
related: https://github.com/FreeCAD/FreeCAD/issues/11676

FEA-eng, 2025-09-14:
I get a bunch of recompute errors when opening this file. It would be good to find a simpler test case.

FEA-eng, 2026-03-24:
[This](https://github.com/FreeCAD/FreeCAD/issues/9725#issuecomment-3289723678) is still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260318 (Git shallow)
Build date: 2026/03/18 23:42:42
Build type: Release
Branch: grafted,grafted
Hash: 7886ebae105b99f176cddbfc62a106ab06a6d021
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

maxwxyz, 2026-03-24:
We can reopen with a simpler working file.

FEA-eng, 2026-10-01:
Needs a simple example, but the issues with drag and drop and moving objects in the tree in general (such as mentioned [here](https://github.com/FreeCAD/FreeCAD/issues/9725#issuecomment-1587997881)) should still be relevant.