FreeCAD/FreeCAD #9726: Unable to read svg on windows with system locale not utf-8

BuddhasPriest, 2023-06-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
When importing svg file on Japanese Windows 11, I get an error because the default system locale is shift-jis (cp932). Saving the svg file in utf-8 also causes an error.
Changing line 1770 of importSVG.py to parser.parse(pythonopen(filename, errors='ignore')) works, but I am not sure if this is a fundamental solution.
```

### Subproject(s) affected?

None

### Problem description

Users cannot import svg files on windows with a system locale other than utf-8.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-06-06:
We need a better system that supports different encodings probably...

Could you try this?

Place this line above line 1770: `import codecs`

Set line 1770 to this: `parser.parse(codecs.open(filename))`

`codecs.open()` is advertised as being more tolerant to different encodings than standard `open()`...

BuddhasPriest, 2023-06-07:
@luzpaz 
Thanks for your comment.
I tried it and still got the codec error...

luzpaz, 2023-06-11:
@BuddhasPriest missing About info  
missing test on dev version

yorikvanhavre, 2023-06-14:
Does the first line of the file (the <xml> tag) informs of the encoding?
Also, what is the error you get when trying to open the file?
I'm not sure using open(errors='ignore')  is a good idea because some people might have encoding errors that will give wrong results...

benj5378, 2023-07-23:
@BuddhasPriest are you able to share your SVG file? Just to prevent false debug results

BuddhasPriest, 2023-07-30:
Dear Yorik van Havre,

I'm  responding GitHub, but will comment.
This code did not work, it fails to load svg created with cp932
codec(Shift-Jis codec).

 I have thought a bit about essential solutions, but currently have no
ideas. I'm stumped.

Sincerely regards,

luzpaz, 2023-08-23:
@BuddhasPriest can you please drag and drop an svg file for testing please?

luzpaz, 2023-09-23:
@BuddhasPriest :point_up_2:

benj5378, 2023-09-27:
> ideas. I'm stumped.

SVG file is not that necessary if we can just make an SVG file that can recreate the bug ourselves (which we very likely can).

luzpaz, 2024-03-14:
> can you please drag and drop an svg file in to this ticket (in order to attach it) for testing please?

@BuddhasPriest

maxwxyz, 2024-08-05:
Issue closed as discussion stalled and we get no further feedback.

closed 2024-08-05 by maxwxyz