FreeCAD/FreeCAD #9729: [Feature Request] how to make Freecad as SCADA?

nex0ma, 2023-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Freecad has an excellent environment for creating a 3D view and techdraws, so we need to discuss the possibilities of organizing data visualization from industrial controllers (signals, graphs, flows, videos, statistics, diagrams). Freecad as SCADA?
```

### Subproject(s) affected?

None

### Problem description

no problem, only idea.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-06:
No problem description.... we don't accept arbitrary requests to mimic other software, explain what problem you have with the current offering and the requirements to have them solved. What features are you missing for what workflow and usecase? be detailed, do the work to create an issue that is actionable.

closed 2023-06-06 by adrianinsaval

nex0ma, 2023-06-06:
ok. I will think. 

The hint about the addon is good. the idea is to place indicators on 3d figures of the production process.

Can I close this ticket for now?

WandererFan, 2023-06-07:
SCADA = System Control and Data Acquisition https://en.wikipedia.org/wiki/SCADA

Possible starting points:
- Evgeniy's schematics workbench: https://forum.freecad.org/viewtopic.php?p=653986#p653986 & https://forum.freecad.org/viewtopic.php?t=73654
- Symbols and Traces PoC: https://github.com/WandererFan/FreeCAD/tree/SymbolsAndTraces

luzpaz, 2023-06-11:
@nex0ma what is your plan? are you going to make an addon or contribute to the above projects :point_up_2:   ?