FreeCAD/FreeCAD #9730: When using "Text Document", object colors disappear and sheet metal attachments can no longer be edited

BobH222, 2023-06-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows Server 2016 Version 1809
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Assembly4 0.50.2
  * Curves 0.6.9
  * fasteners 0.4.56
  * Help 1.0.3
  * MeshRemodel 1.8919.0
  * sheetmetal 0.2.63
  * Trails 2022.1.0
[/code]

and

[code]
OS: Windows Server 2016 Version 1809
Word size of FreeCAD: 64-bit
Version: 0.21.0.32552 (Git)
Build type: Release
Branch: master
Hash: e2b3a5d2f791fb78a61daac9bdab374058f682f2
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * Assembly4 0.50.2
  * Curves 0.6.9
  * fasteners 0.4.56
  * Help 1.0.3
  * MeshRemodel 1.8919.0
  * sheetmetal 0.2.63
  * Trails 2022.1.0
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

1. Create a Sheet metal project.
2. Colorize objects.
3. Create a "Text Document" and input text.
4. Save.
5. Add additional text to the "Text Document".
6. Click on the "X" on the top right of FreeCAD to close the project.
7. You will be prompted to save changes to the document and to the text (I assume). Click Save on both.
8. Reopen the project.
9. Observe that color is lost on all bodies.
10. Observe that the attachment positions of bends can no longer be edited,.

[image]
[image]
[image]
[image]
[image]

Sample project:
[NoSheetMetalEdit.zip](https://github.com/FreeCAD/FreeCAD/files/11658450/NoSheetMetalEdit.zip)

### Anything else?

Discussed in the following Forum Topics:
https://forum.freecad.org/viewtopic.php?p=684071#p684071
https://forum.freecad.org/viewtopic.php?t=78653

I would suggest that if one is embedding a text document into a FreeCAD project, nearly everyone would want, and if not want, would at least accept, that the text is always saved when the project is saved. I would recommend that the double popup be eliminated and the text is just always saved with the project - no option.

Thanks,
Bob.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-07:
CC @shaise (sheetmetal workbench dev)

BobH222, 2023-06-08:
luzpaz,

> This may need to be transferred to sheetmetal workbench repo (https://github.com/shaise/FreeCAD_SheetMetal)

Since the problems only occur when modifying a text document, I assumed it was more to do with the save hierarchy rather than in the Sheet Metal section itself. More likely do do with the Text Document" section but I can't find that in Github. I could use some advice on how to move or link this to the right place.

I also suspect that when the save message is used, the FCStd file is unpacked again but reassembled incorrectly. Something is messing up when it is repacked. It may be at a higher level then either package. That is why I am suggesting the simple fix might be to just always save the text document, and not leave it as an option. (Still, there may be value in funding the root cause, in case there is a lingering bug in either the text document or sheet metal packages that might rear its head again in the future.)

Thanks,
Bob.

luzpaz, 2023-09-05:
CC @shaise

shaise, 2025-01-03:
@luzpaz , @BobH222 
Sorry for this issue slipping under my radar.
Is this the case under ver1.0? I could not reproduce it.

shai

luzpaz, 2025-01-14:
@BobH222 :point_up_2:

luzpaz, 2025-01-14:
I'm going to close this unless bob responds then we can re-open if issue persists.

closed 2025-01-14 by luzpaz