FreeCAD/FreeCAD #9731: [Problem] 3mf Error

1John419, 2023-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Export to 3mf in 0.21.0 causes import error in PrusaSlicer. Same export in 0.20.2 does not produce an error.

Open attached FC file: interlock_flat_notch_4.FCStd.

Select bodies 'male' and 'female' and export to 3mf file.

Attached are export files from 0.20.2 and 0.21.0.

[interlock.zip](https://github.com/FreeCAD/FreeCAD/files/11659585/interlock.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-06:
@wwmayer I think you had worked on the 3mf export feature?

wwmayer, 2023-07-06:
The problem is that with 0.20 all selected objects are converted into a single mesh and this single mesh is written to the 3MF file while with 0.21 several objects are written to several meshes which Prusa apparently doesn't like.

A workaround for now is to first create the meshes using the Mesh wb, merge them to a single mesh and then export it to 3MF.

For v0.22 we can add an option to the preferences to let the user define what to do for the 3MF format.

andesfreedesign, 2023-08-20:
Hello @wwmayer 
Related to this topic, I have found a problem in the current version 0.21 regarding the creation of the thumbnail.png file from the Metadata folder of the exported .3mf files in FreeCAD.
I made a thumbnailer for 3mf files, and when I uploaded it to GitHub it worked...

https://github.com/andesfreedesign/Thumbnailer-3mf

Now checking the exported 3mf file, I found that the thumbnail.png file is created empty...
On another computer I have the FreeCAD_weekly-builds-33345-2023-06-25-conda-Linux-x86_64-py310.AppImage, and I just tried exporting to 3mf and the thumbnailer works perfect. Therefore, some modification in the code, after that date (2023-06-25), must be causing this problem.
I estimate that it may be some modification in the renderer
Thanks in advance for your work
Greetings!

BioSehnsucht, 2023-08-24:
> The problem is that with 0.20 all selected objects are converted into a single mesh and this single mesh is written to the 3MF file while with 0.21 several objects are written to several meshes [...]

In the past I could selectively enable individual bodies  (show/hide them) to affect the output when exporting the part, or alternatively select bodies and export only that body. This statement about how .20 vs .21 handles 3MF implies I should be able to export a single body and have it work, if it was otherwise generating output that Prusa (etc) didin't accept (since there would be only a single mesh output then).

However, that doesn't work - even selecting to output just the singular body (if it's only one) that outputs as a surface still outputs as a surface and not a solid and naturally this fails to load into the slicer. 

Annoyingly, it doesn't even seem to be any recognizable pattern that causes this surface vs solid issue, sometimes a simple sketch with a box inside a box (making a box tube section once Padded) can fail, while a more complicated part might be fine until the 3rd or 4th additive / subtractive object is applied.

gallaghersart, 2023-08-24:
I just installed 0.21 and have same problem with STL I think. Before 0.21 I could highlight a part that had many parts within and export only the part's the were active within main part. Now I just get 3 of the many parts exported winthin. If I fuse them together then export I’m fine. Is this how it will be going forward? or is this something that will get fixed? Also I notice a new options come up when exporting STP and that works. 

I just installed weekly build 0.22.0dev 33981 2023/08/23 17:51:22 on windows 10 b.19041 has same problem. 
Thanks!
`mike

luzpaz, 2023-08-25:
> Related to this topic, I have found a problem in the current version 0.21 regarding the creation of the thumbnail.png file from the Metadata folder of the exported .3mf files in FreeCAD.

@andesfreedesign can you open a new ticket for this ?

andesfreedesign, 2023-08-25:
> > Related to this topic, I have found a problem in the current version 0.21 regarding the creation of the thumbnail.png file from the Metadata folder of the exported .3mf files in FreeCAD.
> 
> @andesfreedesign can you open a new ticket for this ?

@luzpaz 
Yes, of course... I'm going to generate the ticket

wwmayer, 2023-08-30:
> Now checking the exported 3mf file, I found that the thumbnail.png file is created empty...
On another computer 

This could also mean that off-screen rendering doesn't work on this computer. Does it work for you to create an image from an opened 3d view? You will find the command in the Tools menu.
Or if you activate the option to embed a thumbnail into FCStd projects does this work, too?

khimaros, 2023-09-03:
is this a duplicate of https://github.com/FreeCAD/FreeCAD/issues/9731 ?

luzpaz, 2023-09-23:
bumping for activity

wwmayer, 2023-09-24:
The thumbnail issue is handled in a separate issue: https://github.com/FreeCAD/FreeCAD/issues/10364
The problem of this issue is already fixed.
See also : https://github.com/FreeCAD/FreeCAD/issues/10075

closed 2023-09-24 by luzpaz