FreeCAD/FreeCAD #9737: [Problem] Segmentation fault on `freecad --version` command

abdelwahabz, 2023-06-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Manjaro Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.11.3, Qt 5.15.9, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Simply when typing `freecad --version` output:
Python 3.11.3
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x39ab0) [0x7f7d2a64fab0]
#1  /usr/lib/libc.so.6(+0x178e3a) [0x7f7d2a78ee3a]
#2  /usr/lib/libpython3.11.so.1.0(+0x23001e) [0x7f7d2c43001e]
#3  /usr/lib/libpython3.11.so.1.0(+0x2bb20f) [0x7f7d2c4bb20f]
#4  freecad(+0x40bf) [0x55563bf9f0bf]
#5  /usr/lib/libc.so.6(+0x23850) [0x7f7d2a639850]
#6  /usr/lib/libc.so.6(__libc_start_main+0x8a) [0x7f7d2a63990a]
#7  freecad(+0x4a55) [0x55563bf9fa55]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-07:
Please always put crash output in code tags. If you don't then it will spam other unrelated tickets. 

This looks like python (specifically v3.11). Can you recreate this on development version ?

@adrianinsaval are we considering v0.20.x to be compatible with python 3.11 ?

adrianinsaval, 2023-06-07:
Is this your own build or from repo? Does freecadcmd --version work?

Are you running this command in a working x11 session?

adrianinsaval, 2023-06-07:
> @adrianinsaval are we considering v0.20.x to be compatible with python 3.11 ?

0.20.2 should be unless I missed more python 3.11 compatibility fixes on master, this one at least got backported before that release: 639546574e2d4b468f125e0c17d67af73156c9da

adrianinsaval, 2023-06-07:
Arch repo or Manjaro repo? If you experience problems running an arch package on Manjaro you should be expecting occasional problems as it might be compiled against a newer set of dependencies, many times it works without issues but when something happens you must first check that it's not just due to mixing

adrianinsaval, 2023-06-08:
I can reproduce this on archlinux with 0.20 but not with 0.21

adrianinsaval, 2023-06-08:
running `freecadcmd --version` gives a different result though:                                 
```bash
import _frozen_importlib # frozen
import _imp # builtin
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import '_warnings' # <class '_frozen_importlib.BuiltinImporter'>
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
import '_io' # <class '_frozen_importlib.BuiltinImporter'>
import 'marshal' # <class '_frozen_importlib.BuiltinImporter'>
import 'posix' # <class '_frozen_importlib.BuiltinImporter'>
import '_frozen_importlib_external' # <class '_frozen_importlib.FrozenImporter'>
# installing zipimport hook
import 'time' # <class '_frozen_importlib.BuiltinImporter'>
import 'zipimport' # <class '_frozen_importlib.FrozenImporter'>
# installed zipimport hook
# /usr/lib/python3.11/encodings/__pycache__/__init__.cpython-311.pyc matches /usr/lib/python3.11/encodings/__init__.py
# code object from '/usr/lib/python3.11/encodings/__pycache__/__init__.cpython-311.pyc'
import '_codecs' # <class '_frozen_importlib.BuiltinImporter'>
import 'codecs' # <class '_frozen_importlib.FrozenImporter'>
# /usr/lib/python3.11/encodings/__pycache__/aliases.cpython-311.pyc matches /usr/lib/python3.11/encodings/aliases.py
# code object from '/usr/lib/python3.11/encodings/__pycache__/aliases.cpython-311.pyc'
import 'encodings.aliases' # <_frozen_importlib_external.SourceFileLoader object at 0x7f85923ea490>
import 'encodings' # <_frozen_importlib_external.SourceFileLoader object at 0x7f85923de950>
# /usr/lib/python3.11/encodings/__pycache__/utf_8.cpython-311.pyc matches /usr/lib/python3.11/encodings/utf_8.py
# code object from '/usr/lib/python3.11/encodings/__pycache__/utf_8.cpython-311.pyc'
import 'encodings.utf_8' # <_frozen_importlib_external.SourceFileLoader object at 0x7f85923df190>
import '_signal' # <class '_frozen_importlib.BuiltinImporter'>
import '_abc' # <class '_frozen_importlib.BuiltinImporter'>
import 'abc' # <class '_frozen_importlib.FrozenImporter'>
import 'io' # <class '_frozen_importlib.FrozenImporter'>
import '_stat' # <class '_frozen_importlib.BuiltinImporter'>
import 'stat' # <class '_frozen_importlib.FrozenImporter'>
import '_collections_abc' # <class '_frozen_importlib.FrozenImporter'>
import 'genericpath' # <class '_frozen_importlib.FrozenImporter'>
import 'posixpath' # <class '_frozen_importlib.FrozenImporter'>
import 'os' # <class '_frozen_importlib.FrozenImporter'>
import '_sitebuiltins' # <class '_frozen_importlib.FrozenImporter'>
Processing user site-packages
Adding directory: '/home/adrian/.local/lib/python3.11/site-packages'
Processing global site-packages
Adding directory: '/usr/lib/python3.11/site-packages'
Processing .pth file: '/usr/lib/python3.11/site-packages/distutils-precedence.pth'
# /usr/lib/python3.11/site-packages/_distutils_hack/__pycache__/__init__.cpython-311.pyc matches /usr/lib/python3.11/site-packages/_distutils_hack/__init__.py
# code object from '/usr/lib/python3.11/site-packages/_distutils_hack/__pycache__/__init__.cpython-311.pyc'
import '_distutils_hack' # <_frozen_importlib_external.SourceFileLoader object at 0x7f8592220650>
Processing .pth file: '/usr/lib/python3.11/site-packages/matplotlib-3.7.1-py3.11-nspkg.pth'
# /usr/lib/python3.11/__pycache__/types.cpython-311.pyc matches /usr/lib/python3.11/types.py
# code object from '/usr/lib/python3.11/__pycache__/types.cpython-311.pyc'
import 'types' # <_frozen_importlib_external.SourceFileLoader object at 0x7f8592222050>
# /usr/lib/python3.11/importlib/__pycache__/__init__.cpython-311.pyc matches /usr/lib/python3.11/importlib/__init__.py
# code object from '/usr/lib/python3.11/importlib/__pycache__/__init__.cpython-311.pyc'
# /usr/lib/python3.11/__pycache__/warnings.cpython-311.pyc matches /usr/lib/python3.11/warnings.py
# code object from '/usr/lib/python3.11/__pycache__/warnings.cpython-311.pyc'
import 'warnings' # <_frozen_importlib_external.SourceFileLoader object at 0x7f8592223d50>
import 'importlib' # <_frozen_importlib_external.SourceFileLoader object at 0x7f8592223410>
# /usr/lib/python3.11/importlib/__pycache__/_abc.cpython-311.pyc matches /usr/lib/python3.11/importlib/_abc.py
# code object from '/usr/lib/python3.11/importlib/__pycache__/_abc.cpython-311.pyc'
import 'importlib._abc' # <_frozen_importlib_external.SourceFileLoader object at 0x7f8592235fd0>
# /usr/lib/python3.11/__pycache__/contextlib.cpython-311.pyc matches /usr/lib/python3.11/contextlib.py
# code object from '/usr/lib/python3.11/__pycache__/contextlib.cpython-311.pyc'
# /usr/lib/python3.11/collections/__pycache__/__init__.cpython-311.pyc matches /usr/lib/python3.11/collections/__init__.py
# code object from '/usr/lib/python3.11/collections/__pycache__/__init__.cpython-311.pyc'
import 'itertools' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.11/__pycache__/keyword.cpython-311.pyc matches /usr/lib/python3.11/keyword.py
# code object from '/usr/lib/python3.11/__pycache__/keyword.cpython-311.pyc'
import 'keyword' # <_frozen_importlib_external.SourceFileLoader object at 0x7f859225f810>
# /usr/lib/python3.11/__pycache__/operator.cpython-311.pyc matches /usr/lib/python3.11/operator.py
# code object from '/usr/lib/python3.11/__pycache__/operator.cpython-311.pyc'
import '_operator' # <class '_frozen_importlib.BuiltinImporter'>
import 'operator' # <_frozen_importlib_external.SourceFileLoader object at 0x7f8592274310>
# /usr/lib/python3.11/__pycache__/reprlib.cpython-311.pyc matches /usr/lib/python3.11/reprlib.py
[... cut ...]
import 'functools' # <_frozen_importlib_external.SourceFileLoader object at 0x7f859224e150>
import 'contextlib' # <_frozen_importlib_external.SourceFileLoader object at 0x7f8592223990>
import 'importlib.util' # <class '_frozen_importlib.FrozenImporter'>
import 'importlib.machinery' # <class '_frozen_importlib.FrozenImporter'>
# possible namespace for /usr/lib/python3.11/site-packages/mpl_toolkits
import 'site' # <class '_frozen_importlib.FrozenImporter'>
FreeCAD 0.20.2 Revision: 29603 (Git)
```

abdelwahabz, 2023-06-08:
Sory for the missleading, i mean manjaro repo

luzpaz, 2023-06-10:
Devs should decide how if to let v0.20.3 support python 3.11 or not. 

CC @chennes

luzpaz, 2023-09-05:
@a6delwaha6z can you reproduce this on 0.21.1 ?

a6delwaha6z, 2023-09-05:
Hi @luzpaz, with Python 3.11.5, FreeCAD 0.21.0 work fine, even with 0.21.1 release. Thanks

closed 2023-09-05 by adrianinsaval