FreeCAD/FreeCAD #9748: [Problem] PartDesign Body container is a confusing concept (with unfitting name)

adrianinsaval, 2023-06-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
not relevant
```

### Subproject(s) affected?

PartDesign

### Problem description

related to #9747

probably a poor choice of word from early development of PD WB, this name carries unfortunate connotations from common language and from it's use in other software in practice our body container behaves more like what is commonly called a Part in other software, this leads to confusion for many users that start their projects by creating a Part container then get confused when a command asks them to create a body to start modelling, the distinction is not immediately clear. It has also been confusing to some on their usage in assembly WBs (it was initially not supported on asm4 and up to this day some people believe they need to put the body inside a Part to use it).

The solution I favor is of dropping the body container in favor of using a customized Part container, similar to what is described in https://github.com/FreeCAD/FreeCAD/issues/9278#issuecomment-1509899507 for assemblies. The container would then be a variation of the Part container and offer greater flexibility, perhaps with a subgroup containing the cumulative/sequential features and potentially allowing other objects or separate feature histories within the same container (and LCS). A downside of this approach is that it would require a migration script for old files and would break forwards compatibility.

This would have to be evaluated when work on #9747 starts as it would potentially make this concept more confusing which made some prominent members of the community uneasy about such change.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-08:
In another issue I also mentioned another possibility to alleviate this:

> in Part Design the term Body does not fit right and carries some unfortunate expectations from some users (single solid rule) that were needlessly added to the software as limitations, but we can't easily use the Part name because it's already used here... So I'm thinking perhaps we can call the current Part container just Container, that's very generic but still fits IMO.

Might be simpler to implement in software but might invalidate lots of documentation/tutorials and lead to great online confusion. But it's probably a decision we should make before a 1.0 release.

grubertechnic, 2023-06-08:
> in practice our body container behaves more like what is commonly called a Part in other software

No, it is different. The Part container in FreeCAD is that was in other Software used to be a Part. A Part most contains much more as only (psychical) Bodies. Just said that a Part can also have valid more the one Bodies (roller bearings, locking assemblies, shrink discs, compounds, welded parts.....).

It contains often also additional geometry information (axis, points, sections elements, .....), material information, manufacturing information and much more. Also it also can contain drawings (TechDraw), CAM (Path) and calculation (FEM, CFD) information, but that is mostly handled different.

You you link a PartDesign Body to an assembly, you only link a model, but no information. That is unpractical when you want to show or edit any kind of extra information.

Second is when you have Part with more then one PartDesign Bodies, which can also be. Example a welded part, is contains minimum a second Body and welding seams. So you have to link and place (trough transformation) all parts and welding seams separate to the assembly, which makes no sense, since it is one Part. Also then when you want to transform it as Part (what it is). You also have transform all other Bodies, which makes no sense (even it can simplified). But when you try to replace the whole welded Part, you have to delete all links and make new links to the new assembly and transform all again.

I could write all day of the benefits of it, but stop for now. But the Part Container (and PartDesign Body) are really really needed objects and makes as conception really good sense, especially when you work many with (real) Parts, Projects and assemblies.

Maybe it does not make sense for very small Parts and Projects, but in a professional way there is no alternative like it is handled like now.

adrianinsaval, 2023-06-08:
In some ways it looks to me like you are contradicting yourself, see how it is confusing?

> Just said that a Part can also have valid more the one Bodies (roller bearings, locking assemblies, shrink discs, compounds, welded parts.....).

Like I mentioned in the issue this is related to the objective of allowing multiple solids in PD.

> It contains often also additional geometry information (axis, points, sections elements, .....)

A PD body also contains these

> You you link a PartDesign Body to an assembly, you only link a model, but no information. That is unpractical when you want to show or edit any kind of extra information.

What kind of additional information do you want to link? I would say the first statement here is incorrect, when you link to a body everything inside it is exposed on the link, which includes lots of things. The second statement speaks more in favor of a more flexible container (Part) so I don't quite get your point here.

> Second is when you have Part with more then one PartDesign Bodies, which can also be. Example a welded part, is contains minimum a second Body and welding seams. So you have to link and place [...]

what are you on about here? FreeCAD perfectly allows arbitrary levels of nested Parts, using a Part as base container does not make it incompatible with making a more complex Part using several of it even in the same document, besides I mentioned here that this is related with the objective of removing the single solid rule so you can have multiple solids within the same Part anyways, perhaps even with separate feature histories but sharing a common coordinate system (that of the Part) as well as potentially allowing other types of objects.
[image]

In short, I'm looking to expand what is possible to do within PartDesign and the "Body" name is too restrictive, we could just ignore this naming problem and add a bunch of stuff to the body container but the concept is then even more confusing than today.

WandererFan, 2023-06-09:
Body is not a container.  It is a historical record of the steps taken to create the finished object.

We already over-use the heck out of "Part".    Changing "Body" to "Part" doesn't help unless we simultaneously change App::Part to be "Assembly".  

From the Open Assembly Model: "An Assembly is decomposed into sub-assemblies and parts. A Part is the lowest level component which cannot be further decomposed. The diagram shows that an assembly or a subassembly is made up of at least two parts. "

What we call a Body is a often called a "piece part" - A single piece not normally subject to disassembly without destruction or impairment of use, such as resistors, transistors, relays, and gears.
(https://www.dau.edu/glossary/Pages/GlossaryContent.aspx?itemid=28146)

grubertechnic, 2023-06-09:
> In some ways it looks to me like you are contradicting yourself, see how it is confusing?

Where?

> Like I mentioned in the issue this is related to the objective of allowing multiple solids in PD.

A multiple solid PartDesign Body is something different as an Part with multiple PartDesign Bodies. A PartDesign Body to automatic a boolean operation, a Part Container not. So when touching surfaces are included, the result is complete different.

Disclaimer for the examples: it is only considered the cases, where the models are build with PartDesign, not with other WBs.

Example a francis runner where a multiple solid PartDesign Body helps for modeling, but in the Part Container is in the end only one PartDesign Body.

Usually you start to make the runner crown and the runner bottom, since there are first given as parameter. But they are not touching or intersect so it is not allowed atm (singe solid rule). Then you create the blades and pattern them (and do the rest dress up). Then the end result is a singe Part Container with PartDesign Body with a single Solid. But while working, it has a for a certain time a multi solid PartDesign Body. Pretty the similar workflow is while making track wheels, fly wheels, ..... (rim and hub at first, then the rest).

https://forum.freecad.org/viewtopic.php?style=8&t=77004&start=30#p667104

An other example a Part Container with multiple Bodies. Welded parts. They are not boolean operated, but are still one part, since the one position in a BoM, handled while assembling as one Part, bought or manufactured (manufactured with maybe some exceptions) as one part and so on. Also in connection in a PDM or ERP they are handled different. When they are different in the ERP, the change requests are foredoomed. Here is an exapmple, but it still have the singe solid in Part Design limitation. For example i would make all welding seams (or all with the same welding addition) in one multi solid PartDesign Body. A similar build up but complete differt part, a roller bearing. It has multiple PartDesign Bodies, but is only one part (one position in a BoM, handled as one, bought as one, .....).

https://forum.freecad.org/viewtopic.php?style=8&t=56055&start=50#p482639

> > It contains often also additional geometry information (axis, points, sections elements, .....)
> 
> A PD body also contains these

Only when you force it and make workarounds. But usually not. Also it should not be in the PartDesign Body, since it messes it. It still should be in the Part Container (as it works now; i am not complete against it, that it should not be possible to move it in the PartDesign Body, it should be decided user).

> What kind of additional information do you want to link?

For example it is common with a good PDM/ERP combination, that some information are explicid defined and saved in the file (mostly in the Part) that is updated and interchanged in the PDM/ERP.

injection molding:
 - principal direction
 - slider directions
 - 6 supporting points for the definition for measuring condition
 - surfaces (as brep and measured as area) for etching or polish surfaces (for the tools)
 - measuring points (areas)
i have surly forgotten some points, it is a while where i was working for that kind of industries

steel work:
 - surfaces (measured as area, perfect addition as brep) for corrosion protection and its build up (sandblasting, undercoat, between coating, end paints, colors, .....)
 - (welding and/or test (PT, MT, UT) information, but i would do that in the drawing or test plan)

machine construction (dependable on the specialization):
 - transport sizes and masses
 - test flag
 - (without dependencies) sections to the next parts
 - (for me) hydraulic sections and information
 - (for me) HVOF coating volume and material

In your screenshot:
> [image]
> 
> 
when you have a multi Body in there, where starts and ends a PartDesign object? When there are only two pads, i can not say, if it is one or two objects or it is boolean operated or not. This is the reason, why the PartDesign Body was introduced.

I could write much more, but i can not stand the jumping screen of github. Also i find the forum is the better place to discuss that. If you are interested, we can make a jitsi meeting about that. Writing about that is not so easy.

grubertechnic, 2023-06-09:
@WandererFan : perfect explanation.

@adrianinsaval : that the word "Body", especially when it can be multi solid in future, can be misleading is true. Maybe then the word PartDesign item or unit (or Container, but it is no container) would be more appropriate.

adrianinsaval, 2023-06-09:
> Body is not a container. It is a historical record of the steps taken to create the finished object.

It might not be meant to be a container, but it behaves like one. It has it's own coordinate system which you can't reference anything without putting it inside the body, even that something won't be part of the solid. Datum objects and shape binders are more often than not practically independent of the feature history (and some sketches too) so they might as well be outside the feature history tree but still inside the same coordinate system.

> We already over-use the heck out of "Part".

IMO this would be a very appropriate use of Part, more so than the current uses. But if needed I think we can change the name of the current App::Part to Container since it really is a generic container that might be used for anything. We can then have subtypes of Container, such as an Assembly Container or a Part Container which can have slight differences in behaviour such as including only solids in the compound of it's shape for the Part Container.
IMO adapting something that is multi-purpose for multiple uses is not abuse.

> A Part is the lowest level component which cannot be further decomposed

This fits better to what we do in PartDesign than to the current uses of Part so I think you are unknowingly agreeing.

> A PartDesign Body to automatic a boolean operation, a Part Container not.

I am not proposing to use the Part container as is and to dump all PartDesign operations inside it without consideration. The feature history and cumulative nature of PartDesign operations will not be lost. It's just that the feature history will be one level bellow, not directly in the same level as the coordinate system, which can allow adding other things in the same coordinate system that will not be part of the end product. This would make much safer mixing of different workbench features.

> when you have a multi Body in there, where starts and ends a PartDesign object? When there are only two pads, i can not say, if it is one or two objects or it is boolean operated or not. This is the reason, why the PartDesign Body was introduced.

Imagine each of the "Solid" elements in the screenshot has a full feature tree inside them, like the current body but they share the same coordinate system from the Part container instead each having their own

adrianinsaval, 2023-06-14:
I guess something like this would be a better representation of tree structure/hierarchy that I have in mind:
[image]

It is more flexible IMO, by putting the cumulative solids at the same level as the origin object it means we can add all sort of auxiliary stuff on the same coordinate system context so they can be referenced without issues. This can allow the boolean operations in PartDesign to be more useful too.
The Part itself would only include solid objects in it's compound shape

grubertechnic, 2023-06-17:
> It is more flexible IMO,

Really? What if you have to move a Body (or FeatureSet)? Then you are bound on the origin of the part and refactoring is much more work.

Also there is the Body (or FeatureSet) boolean workflow, when you have regions of interest, for example in injection molds (plastic or metal) or bigger castings with region with different sections/usecases. Here a really simple example of clips on a injection molding. It looks maybe at first really wierd, but it is really practically. If i want to move a clip, just move the Body (FeatureSet), all content moves with it, no adaption needed.
[image]

Also if you want you can still refer to the part origin, for example for a draft.
[image]

Also very practical, the origin of the clip is in this example in the section of the counter part. That means you can model the Body (FeatureSet) with the origin in the section, now that it is nice!
[image]

Also a good thing on welded parts, if you have several Bodies (FeatureSet) in it and see, that some Bodies (FeatureSet) must move to an other welding part, because it is better in construction, manufacturing, ..... then you move the whole Body/Bodies (FeatureSet) to the other Part and you have no dependencies to the old part (as long you do not refer something to it). And since the the placement was in the section, you just need to move the Body/Bodies (FeatureSet) to the section coordinate again and you are done. No need to change every sketch, pad, revolution, etc. Do not forget, if there is no origin of the Body (FeatureSet) all sketches/features refer to the origin of the part. This means when new destination the part have an other placement, which is in assemblies 99% the case, you have to edit all sketches/features according to the new origin, that is pretty much work.

Maybe on some people never need this, but for large and/or complex models, were you need structure to get an overview, it is a real lifesaver.

wsteffe, 2023-06-30:
As already emerged in this threed the main addition brought by RT PartDesign model (which I think should be adopted also in FC master) is the possibility to have several branches of history tree (named featuresets by grubetechnic) in the same coordinate system.

This kind of RT Body model: 
- simplifies the crossreferencing between objects in different branches, 
- but it makes more difficult  moving a featureset into a different position.

In the RT branch it is possible (**not obligatory**) to have several  featuresets in the same Body (im the same coordinate system). I am using this option only when the first point is more important then the latter. 

Regarding the _Body_ term (which is the subject of this topic),  I would suggest to replace it with _Geometrical Set_. This term is used in Catia for a generic container that can hold any kind of geometry (multiple solids, faces, shells...). The _Geometrical Set_ of Catia is defined inside a Catia Part (in the same way of a FC Body in a FC Part).

IMO using _Part_ term for current _Body_ ( and a different name for current _Part_) would also be acceptable. This choice would be in better agreement with the step format exported by a current FC Multubody Part. In fact, in the export phase, this kind of part is always converted into an assembly of parts. It may be easily checked reimporting the step file in a commercial CAD  (such as Catia).

ConnorWorrell, 2023-07-02:
A quick thought:

Often times with multi body models (Solidworks) I find the tree too tall and its difficult to tell what bodies are related to what features.

After solving the body, it would be interesting if the operations that constructed the body were displayed under it (duplicating as necessary) The body containers after the first are just for show. Photos below.

Box (1 body)
[image]

Cut through box (2 bodies); Pad on left body.
[image]

Pad to rejoin (1 body)
[image]

sliptonic, 2023-10-13:
I'll run the risk of opening up a can-o-worms again.

The definition of 'Body' seems pretty stable.  That isn't a word we use for many different things around the application.  The singular purpose of the Part Design workbench is to build these things. 

What would be the consequence of renaming the 'Part Design' workbench to 'Body'   not 'Body Design' or something else, just 'Body'?   

We have Arch for designing architecture and Sketcher for making sketches.  TechDraw makes technical drawings and Path designs tool paths. 

Wouldn't a good first step to untangling this mess be to rename Part Design to reflect what it actually does?

JohnOCFII, 2023-10-14:
If that topic is open for discussion, how about generically calling it just "Design?"  In many ways it is the default tool people use to make "Stuff."  It's less about the thing/body/part you are creating, and more about the _act_ of making it.  Just my two cents.

FlachyJoe, 2023-10-14:
My two cents: what about giving to Part the Body capabilities and remove the Body. So we'll have a "stuff" container which can act as functional shape provider (i.e. PartDesign side) and as other shapes container (other workbenches).

WandererFan, 2023-10-14:
Part Design WB ==> Complex Solid WB
Bodies are complex solids vs the simple solids like Cube or Sphere.

sliptonic, 2023-10-18:
> If that topic is open for discussion, how about generically calling it just "Design?" In many ways it is the default tool people use to make "Stuff." 

because in other ways, it isn't the default tool.  Architects do 'design' in Arch.   CNC machinists design toolpaths in Path.

> My two cents: what about giving to Part the Body capabilities and remove the Body. So we'll have a "stuff" container which can act as functional shape provider (i.e. PartDesign side) and as other shapes container (other workbenches).

That's a bigger issue and likely to be controversial.  It might be the right way to go but I'm suggesting a baby-step to start clarifying concepts rather than a technical change to how things work.

> Part Design WB ==> Complex Solid WB
> Bodies are complex solids vs the simple solids like Cube or Sphere.

"Body" is a word we already use and has at least some industry standardization.  "Complex Solid" is introducing a new word to the lexicon.

alpdigitech, 2023-12-20:
When I saw the "Web" workbench it gave off an "amateur" vibe. It should just be an entry in the Help menu which then opens the user's default web browser to the documentation. FreeCAD is the opposite of the linux mantra of tools having one purpose and doing that purpose well. Another case in point, in https://wiki.freecad.org/Draft_Workbench we have this:

> If your primary goal is the production of complex 2D drawings and [DXF](https://wiki.freecad.org/DXF) files, and you don't need 3D modelling, FreeCAD may not be the right choice for you. You may wish to consider a dedicated software program for technical drafting instead, such as [LibreCAD](https://en.wikipedia.org/wiki/LibreCAD) or [QCad](https://en.wikipedia.org/wiki/QCad).

Then why is it there? Either remove it or bury it in the Add-ons.

I'm new to FreeCAD and the first hurdle was actually finding out which one to use. Part or PartDesign. It's so confusing and such a mental barrier to new users. A good programmer writes lots of code. A better programmer deletes lots of code (through refactoring). I hope moving forward Part and PartDesign are merged (hopefully union instead of intersection). Just have an automatic migration path for older documents. Thanks for all the hardworking programmers of FreeCAD, just some honest take on the current valid points of confusion, not meaning to put down any individual.