FreeCAD/FreeCAD #9751: [Problem] Separate sweeps based on helices can't be fused when the profile is circular

FEA-eng, 2023-06-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
[/code]
```

### Subproject(s) affected?

Part

### Problem description

Neither Part Fuse nor Draft Array with Fuse=true can merge separate Part Sweeps based on Part Helixes when the profile is circular. Exporting to step results in multiple separated parts instead of a single continuous solid. Merging works only when the profile is non-circular.

Forum discussion: https://forum.freecad.org/viewtopic.php?t=78879

File:
[Overlapping sweeps on helices.zip](https://github.com/FreeCAD/FreeCAD/files/11707281/Overlapping.sweeps.on.helices.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

edwilliams16, 2023-06-11:
Note in the forum discussion that the failure is scale dependent.  Increase the model size by 10x and it no longer fails.

FEA-eng, 2025-01-26:
Currently, FreeCAD hangs for a while when applying Part Fuse to those two arrays and finally fails with those errors:

```
18:12:13  <Part> ViewProviderExt.cpp(1319): Cannot compute Inventor representation for the shape of Stent_1#Fusion: Bnd_Box is void
18:12:13  <Part> ViewProviderExt.cpp(1319): Cannot compute Inventor representation for the shape of Stent_1#Fusion: Bnd_Box is void
```

FEA-eng, 2025-06-04:
Still relevant

FEA-eng, 2025-08-04:
Still relevant

FEA-eng, 2026-06-06:
Now, recomputing the file attached to this issue makes the model disappear.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260602 (Git shallow)
Build date: 2026/06/02 19:21:10
Build type: Release
Branch: (HEAD detached at 07de249ec)
Hash: 07de249ec749d3fba6710c78da8bca7c66243347
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

FEA-eng, 2026-09-06:
[This](https://github.com/FreeCAD/FreeCAD/issues/9751#issuecomment-4638066443) is still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260720 (Git shallow)
Build date: 2026/07/20 16:51:42
Build type: Release
Branch: grafted,grafted
Hash: 057d51f846f535f4f0b48492e586be0a76b9efe2
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 Compatibility Profile Context 26.5.2.260413/ATI Technologies Inc./AMD Radeon(TM) Graphics
```