FreeCAD/FreeCAD #9762: Strange behavior of shape when using the Draft feature in PartDesign

w035h, 2023-06-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * dodo 1.0.0
  * LCInterlocking
  * Assembly3 0.12.0
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

While designing a part in notices some strange behavior of the PartDesign Draft tool

I first created a part and a sketch as seen below.
[image]

Next I select the to op the resulting pad and click the PartDesign Draft tool and change the angle to 60 degrees. which results in the following part;
[image]

Running check geometry results in the following message.
[image]

I have attached the FreeCAD file as wel for reference. 
[draft-error.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11721875/draft-error.FCStd.zip)

Hope this helps to resolve this. Please let me know if there is more information needed. 

Cheers

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

w035h, 2023-06-12:
Hi @luzpaz, I saw you added the "missing: tested on dev version" tabel so I did a quick test on the latest weekly build. This results in the same errors. Hope this helps. 

[code]
OS: macOS 13.4
Word size of FreeCAD: 64-bit
Version: 0.21.0.33386 (Git)
Build type: Release
Branch: master
Hash: b6e0cfc52b1d33502f72a03c834bb27deaf52d0c
Python 3.11.4, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * dodo 1.0.0
  * LCInterlocking
  * Assembly3 0.12.0
[/code]

luzpaz, 2023-06-13:
@Roy-043 would you mind weighing-in ?

Roy-043, 2023-06-13:
This is not related to the Draft WB.

On the [PartDesign_Draft](https://wiki.freecad.org/PartDesign_Draft) Wiki page it says:

> The Draft tool will only work on faces that are not tangentially connected to other faces.

w035h, 2023-06-14:
I totally missed that in the wiki page sorry for that. Could this be a feature request?

luzpaz, 2023-11-08:
I've updated the ticket to reflect as feature.

ElementW, 2026-09-13:
No change from reported as of 7f2c3c479238b4f01766d8cac955e013ba3a06c8 and OCC 8.1.0.dev1-3d097a0328. Geo is broken in the exact same way.

@maxwxyz I think that's https://github.com/FreeCAD/FreeCAD/labels/3rd%20party%3A%20OCC, and technically a https://github.com/FreeCAD/FreeCAD/labels/Type%3A%20Bug rather than missing feature.
@pieterhijma FYI