FreeCAD/FreeCAD #9767: [ArchSketch+Carbon copy]

jsolersallent, 2023-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * BIM 2021.12.0
  * FreeCAD_SketchArch-master
  * parts_library
  * parts_library.bak
[/code]
```

### Subproject(s) affected?

None

### Problem description

Hi,
I'm trying to make an archsketch and I want to use the carbon copy tool to copy another archsketch in it. The tool works fine with Sketcher, but not with Archsketch.
Thanks

#https://forum.freecad.org/viewtopic.php?t=78881 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-13:
CC @paullee0

jsolersallent, 2023-06-13:
Thanks luzpaz!

luzpaz, 2023-06-13:
Please alter the title of the ticket so it's more elaborate

paullee0, 2023-06-14:
Thanks  @jsolersallent and @luzpaz 

 @jsolersallent, suggested subject  -  "Carbon Copy does not work with SketchObjectPython  (Archsketch)"

paullee0, 2023-06-14:
@abdullahtahiriyo, when you have a gap, see if you would like to have a look.

Forum Discussion  -  https://forum.freecad.org/viewtopic.php?t=78881#p686564

[https://github.com/paullee0/FreeCAD_Ske ... ect.py#L57](https://github.com/paullee0/FreeCAD_SketchArch/blob/master/ArchSketchObject.py#L57)
[https://github.com/paullee0/FreeCAD_Ske ... t.py#L1683](https://github.com/paullee0/FreeCAD_SketchArch/blob/master/ArchSketchObject.py#L1683)

      archSketch = grp.newObject("Sketcher::SketchObjectPython",name)

adrianinsaval, 2023-06-14:
could you provide an example file?

paullee0, 2023-06-14:
@adrianinsaval  File uploaded in forum discussion OP

[Vil·la Savoye.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11747952/Vil.la.Savoye.FCStd.zip)

jsolersallent, 2023-06-14:
@adrianinsaval,  is easy to view.

1. Create a simple Sketch with Archsketch ( it can be fully constrained or not ).
2. Create a new Archsketch and try to create a carbon copy of the other archsketch. The tool is locked.
3. However, if an Archsketch is created and you want to trace a sketch made with Sketcher, it does work.

luzpaz, 2023-06-14:
> @abdullahtahiriyo, when you have a gap, see if you would like to have a look.
> 

JFYI, @paullee0 that abdullah will be AFK for a while (he mentioned in a previous ticket recently) that he won't be available for a period of time (he didn't specify when)

paullee0, 2023-06-14:
> JFYI, @paullee0 that abdullah will be AFK for a while (he mentioned in a previous ticket recently) that he won't be available for a period of time (he didn't specify when)

Thanks for the heads-up, he did mention something similar when he made a simple reply in the forum discussion on this issue :)

> @adrianinsaval, is easy to view.
>
>     1. Create a simple Sketch with Archsketch ( it can be fully constrained or not ).
>
>     2. Create a new Archsketch and try to create a carbon copy of the other [...]

@jsolersallent , thanks for outlining the steps; however,  other peoples may probably not aware that an external add-on is required before it is possible to create the ArchSketch - a Sketcher::SketchObjectPython object.

I have uploaded your file to enable developers to test :D

jsolersallent, 2023-06-14:
> I have uploaded your file to enable developers to test :D

Thanks @paullee0

luzpaz, 2023-10-29:
maybe sketcher devs can revisit this Arch issue ?

Syres916, 2024-07-16:
@paullee0 it will probably have to wait until after 1.0 Release but it looks like I've got a solution https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:CarbonCopy_Fix_SketchArchPython_dev

Just need to do some regression testing and wait for the TNP stuff to calm down but at least some progress.

paullee0, 2024-07-16:
Thanks @Syres916 :)  I have no knowledge about c++, glad you fix it.

closed 2024-12-09 by chennes