FreeCAD/FreeCAD #9770: [Problem] 0.20.2 won't start. "Illegal storage access"

RaNo99, 2023-06-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
The last released version that DOES launch:
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * fasteners 0.4.56
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

Neither 0.21 nor 0.20.2 won't start on my Windows 10. When I'm launching with `--write-log`, it ends up with:

```
Log: All modules with GUIs using InitGui.py are now initialized
Log: Init: Initializing freecad.gears
Log: Init: Initializing freecad.gears... done
Log: All modules with GUIs initialized using pkgutil are now initialized
Log: Init: Loading FreeCAD GUI
Log: Init: Running FreeCADGuiInit.py start script... done
Log: Init: Activating default workbench StartWorkbench
Log: Loading GUI of Web module... done
Log: Loading GUI of Start module... done
Log: Loading Start module... done
Log: Init: Showing main window
Log: Main window restored
Err: Illegal storage access...
Log: Destruct Gui::Application
Err: <Exception> Illegal storage access! Please save your work under a new file name and restart the application!
```

The GUI does not even show. The last released version that launched properly was 0.20.1.

Uninstalling the mods does not help. Changing the initial workbench does not either. Any suggestions on how to proceed?

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-14:
the latest 0.21 version should not have this bug anymore. Your report is very contradictory you put 0.20.2 in the title, 0.21 in the Version field and 0.20.1 in the Full version info field, can you post full version information of exactly which ones are not working?

closed 2023-06-14 by adrianinsaval

RaNo99, 2023-06-14:
Hey, thanks @adrianinsaval . This indeed must be the same issue. The latest version that I referred to was 0.21.0dev-32552 from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds - and this one is still wrong. Do you know where I can download from the Windows installer for 0.21.0.33051 or above?

adrianinsaval, 2023-06-14:
Further down on the same page there is a version for windows available as .7z file, use that one instead. I'll remove the outdated installer to avoid further confusion

RaNo99, 2023-06-14:
The new version 0.21.0.33345 is working fine. Once again, thank you.