FreeCAD/FreeCAD #9772: [Problem] Sketcher indefinitely freezes the application

rabirland, 2023-06-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
  * fasteners 0.4.56
[/code]
```

### Subproject(s) affected?

Sketcher

### Problem description

### Description

The sketch editor can cause the application to go unresponsive indefinitely. The bug is tested in 0.20 release and 0.21 nightly, both produced the same result.

### Repro steps 1 (instant freeze)

1. Open application
2. Go to Sketcher view
3. Create new sketch (on any plane)
4. Add 3 boxes
[image]
5. Create an equality on the side of the middle one and one of the other box
[image]
6. Create an equality with the same side of the middle box and the side of the last box
[image]
7. The application will freeze within ~0.5 to 1 seconds.
[image]

### Repro steps 2 (adding 3 box)
1. Open application
2. Go to sketcher
3. Create new sketch (any plane)
4. Add 3 boxes (same as above)
5. Begin zooming out
6. At certain zoom level, the application will freeze
[image]

#### Extra:

In both Repro 1 and Repro 2 the freeze can be avoided by:
1. After placing a box
2. Close the box tool (important)
3. Zoom out-in
4. Open the box tool again, place another box
5. Close the tool
6. Zoom out-in again
7. Open tool, place more boxes
8. Select the two edges, add equality
9. Zoom out-in
10. Select the other two edges and add equality

Doing the above steps, the application won't freeze but **only** for the current session. Saving the document, reopening it and opening the sketch for edit **will** freeze the application.

### Anything else?

The lack of results in a google search under various terms might show that the issue is limited to a yet unknown source in my PC.
Config:
CPU: AMD Ryzen 7 3800XT (not overclocked)
GPU: Shappire Radeon 5700XT (not overclocked)
MOBO: MSI MAG B550
RAM: Kingston HyperX Fury Black CL16 (32Gb, 1 module, running on 3200MHz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-14:
Please post full version information of your 0.21 version that isn't working, "0.21 nightly" is meaningless first because we don't have nightly builds and second because we don't know what night you downloaded this.

adrianinsaval, 2023-06-14:
Also are you able to reproduce on a clean config? (remove or rename %APPDATA%/FreeCAD)
can you reproduce if you enable software opengl in preferences?

rabirland, 2023-06-14:
> Also are you able to reproduce on a clean config? (remove or rename %APPDATA%/FreeCAD)
can you reproduce if you enable software opengl in preferences? (requires restart of freecad)

Yes, did try an entirely fresh install of 0.20 and it still procuded the issue.

> Please post full version information of your 0.21 version that isn't working, "0.21 nightly" is meaningless first because we don't have nightly builds and second because we don't know what night you downloaded this.

Sorry, wanted to say weekly version. The version I mentioned was from early April. I did a re-test with the actual weekly build (2023-06-11) and it looks like the bug is fixed in this version, the bug can be closed.

closed 2023-06-14 by rabirland