FreeCAD/FreeCAD #9775: Macro keyboard shortcut hint has wrong path

johnsonm, 2023-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Fedora Linux 37 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git) AppImage
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.6
  * sheetmetal 0.2.56
  * ExplodedAssembly
  * fasteners 0.4.52
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * boltsfc 2022.11.5
  * in3dca-freegrid
  * parts_library
  * Defeaturing
  * A2plus 0.4.56a
  * CurvedShapes 1.0.4
  * Manipulator 1.4.9
```

### Subproject(s) affected?

None

### Problem description

It's very helpful that the development build finally gives me a hint as to why for a long time I've been getting an inscrutable warning

"keyboard shortcut(s) 1 2 3 disabled because of conflicts with Std_ViewFront Std
_ViewTop Std_ViewRight respectively."

[image]

Unfortunately, it's misleading. The location is wrong both for 0.20 and 0.21 development.

In 0.20 it's in Preferences → General → Macros ... and in 0.21 it's under Preferences → Python → Macro ...

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-25:
Here's the code reference in master:

https://github.com/FreeCAD/FreeCAD/blob/a3d445e1c9d3706f8ad99fdf8f5902bae1bf3373/src/Gui/Action.cpp#L1060C7-L1062

luzpaz, 2023-06-25:
@adrianinsaval do we want to backport a fix as well ?

adrianinsaval, 2023-06-25:
No we're not doing another 0.20 release

commit f25e765a (2023-06-25): [skip-ci] Fix #9775 - incorrect pref. path on hint
closed 2023-06-25 by adrianinsaval