FreeCAD/FreeCAD #9778: [Problem] Main interface broken when using Font Scaling Factor

hpicelli, 2023-06-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.9, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Main interface os FreeCAD broke when using font scaling factor <1.00
Have noted this issue since 0.18 (ppa, flatpak, deb)

Scaling screen:
[image]

Screen broken (scaling 0.90)
[image]

Screen ok (back to 1.00)
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-22:
IIRC correctly this was fixed in 0.21 already. Please test on the dev release.

hpicelli, 2023-06-23:
Tested with "FreeCAD_weekly-builds-33345-2023-06-22-conda-Linux-x86_64-py310.AppImage" and works OK!!!
Tested with Flatpak FreeCAD Version: 0.20.2.29603 still not working.
When the stable 0.21 will be avaliable?
You can close this issue.
Thank you!

adrianinsaval, 2023-06-23:
> When the stable 0.21 will be avaliable? You can close this issue. Thank you!

in a few weeks likely

closed 2023-06-23 by adrianinsaval