FreeCAD/FreeCAD #9788: [Problem] "Select associated constraints" does not work

YosypC, 2023-06-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
```

### Subproject(s) affected?

None

### Problem description

The command "Select associated constraints" does not work.

https://github.com/FreeCAD/FreeCAD/assets/50027752/957301e0-5f3c-46dd-b504-ac379251b4f5

### Anything else?

[PortaStrumentiLavello.zip](https://github.com/FreeCAD/FreeCAD/files/11778676/PortaStrumentiLavello.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-07-15:
You are selecting a subelement, the endpoint of an element. You have to select the edge of an element.

YosypC, 2023-07-15:
But constraints are also applied to endpoints, not only edges. Selecting edges' constraints selects an unnecessarily big amount of them, how can you tell which one is which? Why is narrowing the selection down with points not a choice?

Roy-043, 2023-07-15:
I just wanted to point out that the command does work.

In the case of the video you are clicking a point where 2 endpoints coincide. It is therefore unclear which point you want. That may be the reason for this behavior.

YosypC, 2024-01-28:
[image]

> In the case of the video you are clicking a point where 2 endpoints coincide. It is therefore unclear which point you want.

Same behavior, only a single point selected. It yields nothing.

closed 2025-01-24 by github-actions[bot]