FreeCAD/FreeCAD #9789: sketcher polyline with undo (cntl-Z) doesn't move point back, but deletes line.

Troncooooo, 2023-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33038 (Git)
Build type: Release
Branch: (HEAD detached at b3b30e9)
Hash: b3b30e951b46e4dbfe0643934a0b27fb59e9bf32
Python 3.10.6, Qt 5.15.9, Coin 4.0.0, Vtk 8.2.0, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Lithophane
  * 3DfindIT 1.2.0
  * EM
  * Alternate_OpenSCAD 1.0.0
  * Curves 0.4.4
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

When I start via part-design - sketcher and use the polyline creation method. After pressing the cntl-z to undo some steps, I'm kept on the original point and not moving a point backwards.

This lead also to strange errors and I got it even crashing.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-18:
confirmed undesired behavior, but don't know how to make it crash.

https://github.com/FreeCAD/FreeCAD/assets/36372335/3a22d8c1-4c63-4c67-8b03-b593712ecaa2

luzpaz, 2023-07-01:
Do we want to fix this for the next release or defer to a minor release ?

adrianinsaval, 2023-07-02:
this isn't a release blocker but if someone submits a fix before release we can consider it for merge in my opinion.

luzpaz, 2023-10-01:
Any volunteers to fix this troubling bug ?

PaddleStroke, 2025-03-13:
Interestingly our bspline tool has 'press F to undo last point' that polyline doesn't have

luzpaz, 2025-06-03:
> Interestingly our bspline tool has 'press F to undo last point' that polyline doesn't have

@PaddleStroke would you mind locating that in the code and link to this ticket ?

luzpaz, 2026-10-10:
@Troncooooo can you still reproduce this on 26.3 RC1 ? https://github.com/FreeCAD/FreeCAD/releases/tag/26.3rc1

Troncooooo, 2026-10-12:
I've tested it. I can reproduce the title. I can't let it crash. The current behavior is that ctrl-z removed the complete polyline, except the current active. My expectation was that it would remove the last added part in the polyline and not the complete polyline.

Troncooooo, 2026-10-12:
I think, I have to add another comment. I've asked the google overlord and it pointed out that using the key R will undo the last segment. I leave this to the designers what is preferred behavior. However I noticed, that pressing the 'M' (mode change) and then doing 'R' won't return the mode to the previous state. This is again a question for the designers if this is intended or not.