FreeCAD/FreeCAD #9792: [Problem] PartDesign tasks fail to launch when body is inside Part and activated

adrianinsaval, 2023-06-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33411 (Git)
Build type: Release
Branch: master
Hash: 127404d793d5c2081383b46a7b677426d9a0c5aa
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods:
```

### Subproject(s) affected?

PartDesign

### Problem description

-new document
-create Part
-create Body, body is automatically placed inside Part however it is not highlighted as activated
(if you try now to create a sketch in this moment it works without problems)
-double click on Body to activate it
-attempt creating a sketch on body (or launch any PD task really), the task is not launched and the error shown in report view

> 14:57:31  <Gui> Document.cpp(385): Invalid sub object 'Unnamed#Part.Body.Box.Body.'

double clicking on the created sketch or feature successfully launchs the task.

### Anything else?

originally reported at https://forum.freecad.org/viewtopic.php?p=688246#p688246

and here the example file provided there (remove .zip extension):
[Yoke.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11782390/Yoke.FCStd.zip)

a3c4f4bb01f5201daf666852444a8f9c5adbda8d is mentioned as introducing this problem. @wwmayer got any ideas?

If we can't fix this before release this commit should be reverted.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-06-21:
The screencast to get an impression.

[2023-06-17 19-21.webm](https://github.com/FreeCAD/FreeCAD/assets/109620914/58e0e4e8-719a-4f46-ae70-120814797be0)

thwerks, 2023-07-01:
whats the last known good build? 33071 is the last one i have that seems to work..

PLChris, 2023-07-01:
Editing a sketch or a feature shows the appropriate panel.

33357 works as well (macOS conda build)

PLChris, 2023-07-01:
> PartDesign
> ### Problem description
> 
> -new document -create Part -create Body, body is automatically placed inside Part however it is not highlighted as activated

To be precise: the body is shown in bold (as it should be when activated), but it is missing the light blue background color.

PLChris, 2023-07-01:
I got two crashes now while testing this after creating a fillet. Anyone else experiencing this too?

[code]
OS: macOS 13.0
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * FC_SU
  * DynamicData 2.46.0
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * dxf-library
  * fcgear 1.0.0
  * fasteners 0.4.54
  * sheetmetal 0.2.62
  * ExplodedAssembly
  * Curves 0.6.8
[/code]

grubertechnic, 2023-07-01:
> whats the last known good build? 33071 is the last one i have that seems to work..

The bisect result was, that it was introduced here a3c4f4bb01f5201daf666852444a8f9c5adbda8d.

> Anyone else experiencing this too?

No, but an other issue. But i do not know, if it is related, but it was occurring around the same time. It freezes when you add a primitive, see https://forum.freecad.org/viewtopic.php?style=8&t=79154. I must do a bisect about this again.

commit c09ef943 (2023-07-09): Fix #9792: PD tasks fail to launch if body is inside Part and activated

introduced by PR #9557
closed 2023-07-09 by 0penBrain