FreeCAD/FreeCAD #9801: [Problem]

RobbeDGreef, 2023-06-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Unknown
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Rocket 3.0.0
  * IconThemes
  * freecad.gears 1.0.0
  * A2plus 0.4.60n
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Problem description

The spreadsheet editor allows the aliases and content of a cell to be changed in the top bar. However, when editing these values enter has to be clicked in order for the changes to apply. This can be annoying when trying to enter multiple aliases in a row that are being copied from somewhere else (e.i. only one hand is on the keyboard, the other is on the mouse).

For this reason, it would be nice if the fields would apply whenever the focus of the field was lost as well. I don't think this is a big change but it could improve usability and speed up the spreadsheet workflow.

Please let me know if I've missed something!

### Anything else?

I have a change for the code that already works, but I think the FreeCAD guidelines say that only PRs that fix previously identified errors should be created, so I figured I would create an issue and only create the PR if this is something that should be changed according to other people.

If I can just add PRs for such small fixes and do not need to create issues for them, let me know for the future :)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

M4xTheMan, 2023-06-27:
Do we need to discuss if people actually like that you've to explicitly confirm a changed Alias? I've to say that I personally like your proposal.

This doesn't change the current behaviour but it might be a good workaround (I use it all the time): https://wiki.freecad.org/Macro_EasyAlias

RobbeDGreef, 2023-06-28:
That macro is pretty interesting thanks, I am definitely going to use that.
I think I am going to go ahead and submit a PR for the change anyway and close this issue since I don't think anyone benefits from having to explicitly confirm as you said.

PR #9836 (2023-07-19): fix: Spreadsheet applies content/alias even when focus lost
The original stylesheet behavior was that the alias or content fields could be edited, but only after enter was pressed the field would actually update. This change makes the field apply when focus is lost on top of when enter is pressed. This makes it easier to enter the alias of a lot of fields at once. 
Closes #9801 

Thank you for creating a pull request to contribute to FreeCAD! Place an "X" in between the brackets below to "check off" to confirm that you have satisfied the requirement, or ask for help in the [FreeCAD forum](http [... cut]
closed 2023-07-19 by chennes