FreeCAD/FreeCAD #9802: [Problem] wire is not closed, but i don't see any open points

colt45s1911a1, 2023-06-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.2
  * CADExchanger
  * CurvedShapes 1.0.4
  * ExtremeProDark 2.7.8
  * fasteners 0.4.56
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

first tim e i thought i did something wrong so i redrew the part, no problem.
next part i drew same thing, i finish drawing and its fully constrained, i go to pad it and i get a warning that the wire is not closed. .
so i redrew the part and no problems it pads fine. I'm not doing anything different.
i have attached both files one that is "broken " and the other one that works. 

### Anything else?

[top side hinge.zip](https://github.com/FreeCAD/FreeCAD/files/11808164/top.side.hinge.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-06-21:
When you use the validate Sketch tool, you get the troublesome vertex. 
[image]

When you look at the Sketch closer, you will see, that there is something slanted.
[image]

Repair that and it works flawless.
[image]

Just note, your Sketch is not good. Tangential connections have no tangential constraint, many overlapping geometry and not well pointed constraints at all. It is in general bad practice and can harm later.

Please close the issue, thanks.

colt45s1911a1, 2023-06-21:
understood thanks.