FreeCAD/FreeCAD #9803: [Problem] The M key cannot recycle through the different modes of polyline tool

raindropsfromsky, 2023-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Actually, I am using Realthunder FreeCAD Link Branch.

Version: 2023.524.0
Revision: 14555 (Git shallow)
Released on: 2023-05-24

I know it is neither 0.20 nor 0.21 versions.
But the Help > Report a bug menu brought me here.
So I hope I am at the right place.
```

### Subproject(s) affected?

None

### Problem description

After drawing a line segment with a Polyline, if I press the M key, it should [cycle through six options](https://wiki.freecad.org/Sketcher_CreatePolyline).
But the Sketcher does not cycle through the different modes.
Instead, the current polyline is terminated and a new polyline is launched.

As a result, the polyline tool can draw just straight line segments in strictly manual mode.
All lacks all the features.

I checked the **Tools > Customize...** menu. 
The M key is properly assigned to the CreatePolyline command.
[image]

(I have not changed this default setting.)

However, I could not help noticing that while the M key is assigned to CreatePolygon command, there is no other key assigned to cycle through the different modes of the Polyline tool.

So, logically speaking, if the M key is supposed to create a new polyline, that is exactly what it is doing right now!
(Even when the current polyline is being drawn, pressing M will terminate the current polyline and start a new one.)

On the whole, some clarity is needed.

### Anything else?

I saw in bug report #6706 that if the Document window is not docked, some FreeCAD commands don't work. That bug specifically mentions that M key stops working in polyline. 

But in my case, the Document window is docked, so this factor does not apply.
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-22:
link branch bugs must be reported to realthunder's repo. We do not deal with them here. The link you followed is likely just something realthunder didn't bother changing. You may reopen the issue if you are able to reproduce this on master branch 0.21

closed 2023-06-22 by adrianinsaval

luzpaz, 2023-06-23:
@raindropsfromsky report Link bugs to https://github.com/realthunder/FreeCAD