FreeCAD/FreeCAD #9804: Re-importing python module fails with ImportError: 'FreeCAD' is not a built-in module

voneiden, 2023-06-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Gentoo Linux (sway)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git) AppImage
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

Note: not reloading but attempting to remove the module  and re-importing fails. 

```python

import sys
import FreeCAD
del sys.modules["FreeCAD"]
import FreeCAD

ImportError: 'FreeCAD' is not a built-in module
```

### Anything else?

It seems to land here for some reason: https://github.com/python/cpython/blob/main/Lib/importlib/_bootstrap.py#L415

LinuxCNC had a similar error and they fixed it here: https://github.com/LinuxCNC/linuxcnc/issues/825

The use case is with [ocp-freecad-cam](https://github.com/voneiden/ocp-freecad-cam) which depends on FreeCAD. The toolpaths can be visualized for example using [CQ-Editor](https://github.com/CadQuery/CQ-editor), which has a feature that reloads all modules imported by the script that is being rendered, and with FreeCAD present it breaks down as described above. 

I'm able to work around by caching the FreeCAD import in package __init__, so in that regard the priority for this is trivial.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-26:
@wwmayer would you have a moment to take a look ?

adrianinsaval, 2023-06-29:
is that even the correct way of doing this?

voneiden, 2023-06-29:
@adrianinsaval manipulating sys.modules is explicitly documented at https://docs.python.org/3/library/sys.html#sys.modules

> This is a dictionary that maps module names to modules which have already been loaded. This can be manipulated to force reloading of modules and other tricks.

adrianinsaval, 2023-07-10:
can you explain better how you are running into this? How was FreeCAD added to your python search path? It might be necessary to remove all FreeCAD related modules before trying to import again

Also, @luzpaz I'm not sure if it's worth putting this in the 0.21 project, it doesn't seem critical.

voneiden, 2023-07-10:
> it doesn't seem critical.

Yep, definitely trivial priority.

> can you explain better how you are running into this?

I use an extracted appimage of freecad and the search paths are added via a pth file. I've detailed the setup here: https://github.com/voneiden/ocp-freecad-cam#using-freecad-appimage

In practice I ran into this issue with how [CQ-Editor](https://github.com/CadQuery/CQ-editor) executes [cadquery](https://github.com/CadQuery/cadquery) files: it has a context manager that notes all the loaded modules before executing the file and removes all new modules using the method above after the execution. So regarding

> It might be necessary to remove all FreeCAD related modules before trying to import again

I believe in this case all FreeCAD related modules are be removed by the context manager.

However without the [workaround](https://github.com/voneiden/ocp-freecad-cam/commit/48938704f2e34912dc5c92c4625253bf33524733) (wee bit hacky caching of the FreeCAD module into sys), subsequent re-execution will fail with the same `ImportError: 'FreeCAD' is not a built-in module`.

I'm not qualified to determine if this is normal or abnormal behaviour for a Python C module, but I can confirm that for example zlib does not exhibit this behaviour:

```python
>>> import zlib
>>> reload(zlib)
<module 'zlib' from '/home/voneiden/miniconda3/envs/cq/lib/python3.10/lib-dynload/zlib.cpython-310-x86_64-linux-gnu.so'>
>>> del sys.modules["zlib"]
>>> import zlib
```

same compared to FreeCAD:

```python
>>> import FreeCAD
>>> reload(FreeCAD)
<module 'FreeCAD' (built-in)>
>>> del sys.modules["FreeCAD"]
>>> import FreeCAD
 ImportError: 'FreeCAD' is not a built-in module
```

What catches my eye is that while both are C modules, FreeCAD repots as a "built-in" when reloaded whereas zlib does not.

wwmayer, 2023-07-11:
Do you use the FreeCAD or FreeCADCmd executable or start the Python3 interpreter and import FreeCAD as module? The problem you describe only happens for the latter case.

voneiden, 2023-07-11:
The latter, yes. Maybe in the former case the module doesn't get garbage collected?

wwmayer, 2023-07-12:
> Maybe in the former case the module doesn't get garbage collected?

It can only be garbage collected when its reference counter reaches zero. But this is very unlikely to happen and won't be achieved by removing it from the sys.modules dict.

The reason that it fails when loading FreeCAD as extension module (i.e. loading FreeCAD.so) is that it additionally may be created as built-in module. In src/App/Application.cpp there is a function init_freecad_module() that is used twice. 
When used in Application::setupPythonTypes() then beforehand it checks whether a FreeCAD module exists or not but in Application::initConfig() there is no such check.

So, this could eventually be the reason of this issue.

bgbsww, 2023-12-12:
So, this issue appears to be caused because Application::initConfig calls PyImport_AppendInittab to add "FreeCAD" and "__FreeCADBase__" to the INITTAB used in cpython ( https://github.com/python/cpython/blob/7316dfb0ebc46aedf484c1f15f03a0a309d12a42/Python/import.c#L1482 ).

Once that entry is in there, it's game over because there is no provision to remove it short of completely shutting down the python interpreter.  Unless we implement some alternate approach to getting the permanent objects inserted in python, this can't be solved with the LinuxCNC patch referred to.

If the module isn't found, is_builtin("FreeCAD") is always going to return true, causing python to go look for a builtin that isn't there are raise the error shown.

Note that Application::initConfig() is not called a second time, because the search for an import already failed long before that point.

This seems like major surgery for a fringe case that has a workaround, I would suggest we close this and leave things alone.   ocp-freecad-cam is cool, I'm certainly not denigrating it.

voneiden, 2023-12-13:
Sure, sounds absolutely reasonable to close this as there is no simple solution. Thanks everyone for investigating and documenting the cause.

closed 2023-12-13 by voneiden