FreeCAD/FreeCAD #9806: [Problem] View rotation only possible along one axis if rotated using the navigation cube's arrows using Turntable

YosypC, 2023-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
```

### Subproject(s) affected?

None

### Problem description

https://github.com/FreeCAD/FreeCAD/assets/50027752/b357c4bf-cea0-4293-adb4-6a7e4a430111

The problem does not occur with trackball.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-06-29:
Can not reproduce that.

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.9.2, Qt 5.15.2, Coin 4.0.0, Vtk 9.0.1, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * QuickMeasure 2022.10.28
  * Assembly4 0.12.6
  * CurvedShapes 1.0.4
  * Curves 0.6.8
```

Rexbas, 2023-06-30:
I can reproduce. It happens when I look at the front, right, left or rear when the view is rotated + or - 90 degrees. Also happens with free turntable.
[image]
[image]

Rexbas, 2023-06-30:
I tried the same in Blender and Fusion 360. Blender seems to have the same problem although the behavior is a little bit different. In Fusion 360 the camera modes are called free and constrained but I believe it is the same as trackball and turntable respectively. I suppose the camera pose that we try to use is beyond the limit of the constrained/turntable mode. Fusion 360 kinda solves this by resetting the camera to a pose that is within the restrictions of the turntable mode (see attached video).

Basically the navigation cube could force the camera to a pose that is beyond the constraints of the turntable mode. I see two possible solutions, although I am not sure if they could be called solutions. The first option is to manually change the camera pose to a pose that does not have problems by interacting with the navigation cube. For the second option we could try to do something similar as Fusion does but some people might see this as a bug too.

I am not sure what the best solution is.

https://github.com/FreeCAD/FreeCAD/assets/14298143/8e74831d-4a2e-436b-b865-567a8ff35db0

grubertechnic, 2023-06-30:
Still can not reproduce that, in CAD and Blender mouse setting.

Rexbas, 2023-06-30:
That is strange, are you sure you have selected turntable orbit style and oriented the camera in the same way as I have posted previously. Trackball orbit style works just fine. It happens in all navigation styles and even other programs. This problem also happens in Blender, Fusion 360 and Onshape.

grubertechnic, 2023-06-30:
Oh, that i was missing, yes now i can reproduce that. But is that not the intention to rotate around the z axis of turntable?

Rexbas, 2023-06-30:
>But is that not the intention to rotate around the z axis of turntable?

I believe that is true.

grubertechnic, 2023-06-30:
I mean, still a little bit weird, but that must someone confirm, that have experience in the turntable property. It was just guessing from me (i do not use it).

Jolbas, 2023-09-11:
I can think of two possible ways to handle this. One is to modify the behaviour when using the mouse orbit. As it is working now when z axis is vertical but behave as if you are leaning your head when the z axis is not vertical. The other is to prohibit z axis to deviate from screen vertical.

closed 2025-02-03 by github-actions[bot]

FEA-eng, 2025-06-08:
Still relevant

luzpaz, 2025-06-08:
@Rexbas what's the next step, pose your observations in https://github.com/FreeCAD/FreeCAD/issues/9806#issuecomment-1615007126 to the working group or design group ?

luzpaz, 2025-06-09:
@FreeCAD/design-working-group would you mind weighing-in ?

MisterMakerNL, 2025-08-16:
I think atm, with 3k issues and not so many funding in contrast of the size of the issues/codebase we are just trying to fix the most annoying things. 
It has to be annoying enough that a dev gets annoyed by it, or big enough that it stops someone from using the program.
This feels like slightly annoying but not annoying enough for me.
But don't get me wrong some users find other things annoying. Like I also made a bunch of those 3k issues/feature requests that ended up in the void. So I understand that it sucks that it's not getting the attention you would like it too have it.l, sadly that's where we are atm.

luzpaz, 2025-08-16:
@ssteinerx relax there friend, you keep getting upset by stuff like that you're going to burn out. This is a free libre opensource project that consists of many volunteers. There is an association but still in its infancy, and it's starting to get more organized to improve the the bug fixing process. We're still tweaking the bot but it helps to shake the tree every so often.

ssteinerx, 2025-08-19:
@luzpaz -- not all that upset, really, but lately I've had a few auto-closes turn into super-mega-crashes and the "close-bots" seem to be getting more aggressive these days.  

Hey...wait...is anthropomorphizing the "close-bots", and ascribing ulterior motives thereto a sign of burn-out?  

Uh oh... 😂