FreeCAD/FreeCAD #9809: Style gets reset when deleting an item

pvginkel, 2023-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods: 
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Draft

### Problem description

I can't reproduce this in a new drawing, but I know I've had this in multiple drawings.

Steps to reproduce:

- Switch to Draft.
- Select a rectangle in the file, e.g. the first one, Rectangle059.
- Change the Draw Style to Dashed. The rectangle does become dashed.
- Delete an item, e.g. Dimension030 just above it.
- The style of the rectangle gets reset to solid.

I'm running on Windows 11 and the latest stable build. I also rested this with the latest build which came out today: FreeCAD_weekly-builds-33345-2023-06-22-conda-Windows-x86_64-py310.7z.

[FreeCAD reproduction.zip](https://github.com/FreeCAD/FreeCAD/files/11839908/FreeCAD.reproduction.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-06-23:
I don't class this as a bug, it's a user education issue. Layers will always/should take priority. If you want to change the Draw Style then create a new layer with the Dashed draw style and move the object to the correct layer.

pvginkel, 2023-06-23:
Thank you. I searched on Google, the wiki and the forum and I couldn't find any reference to this. Thank you for explaining this though. I do like FreeCAD and this was the only thing that was really bugging me.

closed 2023-06-23 by pvginkel