FreeCAD/FreeCAD #9817: Legacy Python DXF importer not working

transistorgrab, 2023-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4
  * kicadStepUpMod 10.10.4
[/code]
```

### Subproject(s) affected?

None

### Problem description

When trying to import DXF using the legacy Python importer only generates an error:

13:00:03  DXF libraries need to be updated. Trying to download...
13:00:03  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfColorMap.py ...
13:00:03  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfImportObjects.py ...
13:00:03  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfLibrary.py ...
13:00:03  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfReader.py ...
13:00:03  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "D:\Program Files\FreeCAD 0.20\Mod\Draft\importDXF.py", line 2827, in insert
    getDXFlibs()
  File "D:\Program Files\FreeCAD 0.20\Mod\Draft\importDXF.py", line 202, in getDXFlibs
    import dxfColorMap, dxfLibrary, dxfReader
  File "D:\Program Files\FreeCAD 0.20\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
<class 'SyntaxError'>: invalid syntax (dxfReader.py, line 88)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-06-29:
please test on 0.21

transistorgrab, 2023-06-30:
Tested right now, looks the same:

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4 0.50.2
  * kicadStepUpMod 10.19.4
```

```
09:39:36  DXF libraries need to be updated. Trying to download...
09:39:36  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfColorMap.py ...
09:39:36  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfImportObjects.py ...
09:39:36  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfLibrary.py ...
09:39:36  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.40/dxfReader.py ...
09:39:36  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "D:\Program Files\FreeCAD_DEV\FreeCAD_weekly-builds-33458-2023-06-30-conda-Windows-x86_64-py310\Mod\Draft\importDXF.py", line 2762, in insert
    getDXFlibs()
  File "D:\Program Files\FreeCAD_DEV\FreeCAD_weekly-builds-33458-2023-06-30-conda-Windows-x86_64-py310\Mod\Draft\importDXF.py", line 205, in getDXFlibs
    import dxfColorMap, dxfLibrary, dxfReader
  File "D:\Program Files\FreeCAD_DEV\FreeCAD_weekly-builds-33458-2023-06-30-conda-Windows-x86_64-py310\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
<class 'SyntaxError'>: invalid syntax (dxfReader.py, line 88)
```

yorikvanhavre, 2023-07-04:
Should be fixed with the above commit.
In the meantime, you can manually install the dxf libraries from the addon manager, that should do the trick...

closed 2023-07-05 by yorikvanhavre

transistorgrab, 2023-07-05:
I installed the plugin as suggested: this also fails reading the DXF with this message:
18:24:19  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "D:\Program Files\FreeCAD 0.20\Mod\Draft\importDXF.py", line 2837, in insert
    processdxf(doc, filename)
  File "D:\Program Files\FreeCAD 0.20\Mod\Draft\importDXF.py", line 2191, in processdxf
    drawing = dxfReader.readDXF(filename)
  File "C:\Users\andre\AppData\Roaming\FreeCAD\Mod\dxf-library\dxfReader.py", line 399, in readDXF
    (infile, drawing) = sm.run((infile, None))
<class 'TypeError'>: cannot unpack non-iterable bool object

luzpaz, 2023-07-05:
@transistorgrab the fix was merged in to development build and wasnt backported to v0.20.x (which looks like you're using). To test it you need to wait for the next windows dev build to generate.

yorikvanhavre, 2023-07-10:
There is no much sense in backporting anyway, since 0.21 is around the corner, there will be no more update of 0.20...

DeepCoreSystem, 2023-08-28:
@yorikvanhavre 
There are [more cases](https://github.com/easyw/kicadStepUpMod/issues/181#issuecomment-1695225728) that are affected by some quirk with the legacy DXF importer.

Btw. I don't understand why simply bumping the version number (#9873) should fix something ...

yorikvanhavre, 2023-08-28:
> Btw. I don't understand why simply bumping the version number (https://github.com/FreeCAD/FreeCAD/pull/9873) should fix something ...

Because FreeCAD will fetch a specific branch of the importer based on that number... As the number had not been updated, the latest fixes had not been pulled in. That whole system was needed at a certain time when stable and development versions of FreeCAD needed different dxflib versions. This is no longer the case, though, we could retire that system

> There are https://github.com/easyw/kicadStepUpMod/issues/181#issuecomment-1695225728 that are affected by some quirk with the legacy DXF importer.

I'm trying the files now, it seems like indeed the legacy importer has some problem with the encoding, but the non-legacy one seems to open it fine:

[image]

DeepCoreSystem, 2023-08-28:
@yorikvanhavre Thank you for the explanation!

Based on that, I forced FreeCAD 0.21 to use the V1.40 of the legacy importer, and ... it worked !!

easyw, 2023-08-28:
> I'm trying the files now, it seems like indeed the legacy importer has some problem with the encoding

@yorikvanhavre 
it would be possible to fix also the legacy importer?
It has some features useful for my kicadStepUp WB

yorikvanhavre, 2023-08-29:
It's complicated because it's deeper into one of the external libs used by our code... But I'll have a look
Unfortunately I haven't found a way to change the encoding of a DXF file... That's a pity

DeepCoreSystem, 2023-08-29:
@yorikvanhavre 
> Unfortunately I haven't found a way to change the encoding of a DXF file...

What encoding do you mean? ASCII vs. UTF-8 ?

yorikvanhavre, 2023-09-01:
Sorry, my bad, I mixed issues... There is no encoding issue here

easyw, 2023-09-06:
it seems the issue is related to the absence of the acad version in the dxf file... this was just fine in previous releases of the dxfImportLib
https://github.com/yorikvanhavre/Draft-dxf-importer/blob/25931033397a4bdfbf62cdca4393609022752f57/dxfReader.py#L259

yorikvanhavre, 2023-09-08:
That should be fixed with https://github.com/yorikvanhavre/Draft-dxf-importer/pull/26

Pugemon, 2024-06-27:
> That should be fixed with [yorikvanhavre/Draft-dxf-importer#26](https://github.com/yorikvanhavre/Draft-dxf-importer/pull/26)

No, it's not fixed 
I'm getting same error

```
19:07:20  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.41/dxfColorMap.py ...
19:07:20  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.41/dxfImportObjects.py ...
19:07:20  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.41/dxfLibrary.py ...
19:07:21  downloading https://raw.githubusercontent.com/yorikvanhavre/Draft-dxf-importer/1.41/dxfReader.py ...
19:07:21  opening C:\Users\Pugemon\AppData\Local\Temp\fc-03448-440688-000002.dxf...
19:07:21  There has been an error:
19:07:21  First section is not HEADER
19:07:21  Traceback (most recent call last):
19:07:21    File "C:\Users\Pugemon\Downloads\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\Mod\OpenSCAD\OpenSCADCommands.py", line 407, in addelement
19:07:21      importCSG.insert(tmpfilename,doc.Name)
19:07:21    File "C:\Users\Pugemon\Downloads\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\Mod\OpenSCAD\importCSG.py", line 154, in insert
19:07:21      processcsg(filename)
19:07:21    File "C:\Users\Pugemon\Downloads\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\Mod\OpenSCAD\importCSG.py", line 179, in processcsg
19:07:21      result = parser.parse(f.read())
19:07:21    File "C:\Users\Pugemon\Downloads\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\bin\lib\site-packages\ply\yacc.py", line 333, in parse
19:07:21      return self.parseopt_notrack(input, lexer, debug, tracking, tokenfunc)
19:07:21    File "C:\Users\Pugemon\Downloads\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\bin\lib\site-packages\ply\yacc.py", line 1120, in parseopt_notrack
19:07:21      p.callable(pslice)
19:07:21    File "C:\Users\Pugemon\Downloads\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\Mod\OpenSCAD\importCSG.py", line 1311, in p_text_action
19:07:21      p[0] = [processTextCmd(t)]
19:07:21    File "C:\Users\Pugemon\Downloads\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\Mod\OpenSCAD\importCSG.py", line 952, in processTextCmd
19:07:21      face = importDXFface(tmpfilename,None,None)
19:07:21    File "C:\Users\Pugemon\Downloads\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\Mod\OpenSCAD\OpenSCAD2Dgeom.py", line 486, in importDXFface
19:07:21      layers = importDXF.processdxf(doc,filename,False,False) or importDXF.layers
19:07:21    File "C:\Users\Pugemon\Downloads\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\Mod\Draft\importDXF.py", line 2147, in processdxf
19:07:21      drawing = dxfReader.readDXF(filename)
19:07:21    File "C:\Users\Pugemon\AppData\Roaming\FreeCAD\Macro\dxfReader.py", line 399, in readDXF
19:07:21      (infile, drawing) = sm.run((infile, None))
19:07:21  TypeError: cannot unpack non-iterable bool object
```

DeepCoreSystem, 2024-06-28:
@Pugemon 
It is not the same error, because your log states _First section is not HEADER_

There must be a problem with the DXF file that it is missing the key _HEADER_

Post the first 20-30 lines of your DXF file ...

commit 6b588a8e (2023-07-05): Draft: Bumped dxflib version - fixes #9817