FreeCAD/FreeCAD #9818: [Feature Request] Expose UI Color settings to the user

NXTler, 2023-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.11.3, Qt 5.15.9, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * A2plus 0.4.60n
  * QuickMeasure 2022.10.28
  * Curves 0.6.12
  * CfdOF 1.23.1
  * TabBar
  * IconThemes
  * Glass
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Core

### Problem description

Hello all,
The colors of any UI can drastically change the feel and usability of it, and every person prefers different colors for their personal usage; hence, I think it's very important that the colors are easily changeable.
In FreeCAD version 0.20 the process of changing the UI's colors is quite tedious. Custom .css files have to be created, which are too long and complicated for the average user, that just want to change the tree's background for example.
More color options should be exposed through the settings, like it has been already done for the start menu.

What are your thoughts about this?

### Anything else?

Here is a screenshot taken from the settings of Solid Edge as an example:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-06-28:
This is a more complex issue than it seems because since there are many ways to set colors in a Qt app, care must be taken with what overriding what. But this seems like a good idea, in case no css style is used, an easy, default one could be done with a simple dialog like this

adrianinsaval, 2023-06-29:
I don't see a compelling use case for fine control of every single UI element like the tree background, stuff like the colors in sketcher I understand and those are therefore configurable, stuff like this is just more maintenance burden to satisfy a niche whim from a fraction of users so I would like to close this issue.

We already have trouble now because some colors are set using preferences, others using stylesheets, and a few are burried in code, adding more to the mix is not going to help.

Consider developing this as an addon instead.

luzpaz, 2023-07-09:
I think finetuning as an option in FreeCAD core is great. It will make us consider a better approach to standardizing how we set colors (and organize the hacks that have been made to FC in the past on this issue).

adrianinsaval, 2023-07-09:
I disagree, I think this would just be adding more unnecessary complexity to the pile and make it even harder to make good themes. More colours as preferences is undesirable IMO.

yorikvanhavre, 2023-07-10:
My idea was simply a way to change, say, the 5 or 6 basic standard them colors (window fore/background, selected fore/background, etc). Indeed it's not enough for proper theming, but it could be useful to some.

Note that this can 100% be done in an addon. No need to be part of Core. Such an addon could simply offer a preference page to set the colors, build a qss from it, and apply that qss...

MisterMakerNL, 2023-07-12:
I can remember when I was a young boy we had windows 3.1 and we could change all colors.
One day I changed everything to black on the school computer.
I don't think they liked that.
I also don't think giving normal users these powers.

adrianinsaval, 2023-07-12:
An add-on for this would be a perfect solution. For the core I think we should reject this request

And I don't think the issue tracker is for add-on requests

luzpaz, 2023-07-13:
Ok, I concede. Addon is a fair compromise. I'll transfer this to Addons repo