FreeCAD/FreeCAD #9819: [Feature Request] Better Icon alignment

NXTler, 2023-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.11.3, Qt 5.15.9, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * PieMenu 1.2.7
  * sheetmetal 0.2.63
  * toSketch 1.0.1
  * A2plus 0.4.60n
  * QuickMeasure 2022.10.28
  * Curves 0.6.12
  * CfdOF 1.23.1
  * TabBar
  * IconThemes
  * Glass
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Core

### Problem description

Hello all,
when placing the toolbars on the left or right, the Icons will not align in a straight line:
[image]

### Anything else?

Other Programs handle this by aligning all Icons to the left side, not the Center:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-25:
If you're going to ask for UI/UX improvements, please use the development version v0.21.x  
TIA

NXTler, 2023-06-27:
Though this is still present in version v.0.21.0

yorikvanhavre, 2023-06-28:
Possible solutions https://forum.qt.io/topic/59049/left-align-qtoolbuttons-in-a-vertical-qtoolbar

adrianinsaval, 2023-06-29:
> Possible solutions https://forum.qt.io/topic/59049/left-align-qtoolbuttons-in-a-vertical-qtoolbar

if applying something like this care must be taken to not negatively affect right to left systems.

How does it look when you use the appimage or enable a stylesheet? this might rather be a problem with the system theme you are using. Are you running in a gtk system that uses a qt theme dervied from the gtk theme? those are usually buggy. Although it kinda looks like breeze too.

NXTler, 2023-06-29:
I'm using KDE with breeze. Using the release version or the stylesheets changes nothing about the icon alignment.

adrianinsaval, 2023-06-29:
And the appimage? (The appimage doesn't load the system theme)

Can you try configuring a different style on your KDE settings? Like oxygen or fusion

NXTler, 2023-06-29:
Yeah, I'm running the Appimage right now. Changing the KDE style does not change the Icon behavior.

luzpaz, 2023-07-01:
@MisterMakerNL would you mind weighing-in ? TIA!

MisterMakerNL, 2023-07-01:
It's fixed in Linkstage and in my theme pr. Also in ExtremeProDark preference pack.

luzpaz, 2023-07-01:
@MisterMakerNL Thanks! One more request, can you permlink to the stylesheet code that does this.perhaps ?

MisterMakerNL, 2023-07-02:
Currently not near my computer but you can see how it's fixed with the code here. https://github.com/MisterMakerNL/Freecad-Built-in-themes-beta

MisterMakerNL, 2023-07-12:
In the new stylesheets it looks like this:
[image]
It's added here: https://github.com/FreeCAD/FreeCAD/pull/9715
Although still not added in the no-stylesheet. But there can probably be a discussion that the no-stylesheet should be as universal as possible and the old way is maybe a bit more easy to click on since the arrow is located next to the icon instead in it?

luzpaz, 2023-07-13:
Since the fix is available in the new addon, what should we do with this ticket ?

NXTler, 2023-08-07:
This is exactly what I was looking for! I don't think it affects the usability that much and looks a bit more polished, so I don's see a reason why not to add it to no-stylesheet.

MisterMakerNL, 2023-08-07:
> Since the fix is available in the new addon, what should we do with this ticket ?

This is already in 0.21 so can be closed in my opinion.

closed 2023-08-07 by NXTler