FreeCAD/FreeCAD #9823: [Problem] STL, 3MF, etc export fails on complex object with B-splines

hpax, 2023-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Fedora Linux 38 (Workstation Edition) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33447 (Git)
Build type: Release
Branch: master
Hash: 090ae4571a589171dccb5dcabaf9d71e453ab2f3
Python 3.11.3, Qt 5.15.10, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

The attached FreeCAD file renders correctly in all view modes in the GUI, but when exporting it to STL, 3MF, AMF or OBJ (all the formats I tried) only the first subtractive pipe (a straight edge, without B-splines) appears in the output. This is tested both with Cura and by importing the files into FreeCAD itself.

Unlike issue 9731, this is *not* a regression; the behavior is equally reproducible on FreeCAD 0.20.2 as shipped with Fedora Linux 38 (freecad-0.20.2-3.fc38.x86_64.rpm).

### Anything else?

Test case input file:
[testcase-fcstd.zip](https://github.com/FreeCAD/FreeCAD/files/11861506/testcase-fcstd.zip)

Test case output files:
[testcase-output.zip](https://github.com/FreeCAD/FreeCAD/files/11861507/testcase-output.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-06-26:
You messed up somewhere at _Channel cross002_. But you **still can** export the correct file. Select _SubtractivePipe004_ and press _File → Export..._

See the screenshot below.

<details><summary>Screenshot</summary>

[image]

</details>

luzpaz, 2023-06-26:
Closing due to user error.

hpax, 2023-06-27:
I wanted to confirm that I was able to track down the user error -- I had missed a recompute red mark. Regenerating the flagged operation did, indeed, correct the problem.

I apologize for the noise, and thank you for FreeCAD - it is a great program.