FreeCAD/FreeCAD #9824: [Problem] Using formula editor in pad parameters disallows further non-formula parameter modification

hairhorn, 2023-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Modification of the Pad parameter "length" using the formula editor seems to disallow future non-formula modifications to this parameter value. No issues present when only using the length spin box to modify the value.

Steps to recreate:
1. Create Sketch
2. Apply Pad to that Sketch.
3. Double click on the Pad in the model tree.
4. Click on formula editor in the Length spin box.
5. Type any value (or retain pre-populated value) and click OK in the formula editor to apply. (Note: upon returning to the Pad Parameters / Tasks pane, the Length spin box is greyed out and does not accept input - as far as I can tell this is normal behavior immediately after loading a formula)
6. Click Ok in Pad Parameters / Tasks pane to return to the model tree and apply change.
7. Double click on the Pad in the model tree. The spin box is still greyed out and does not allow changes via typing or via the spin box buttons. Use of the formula editor is still allowed and will result in an update.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-06-29:
> Is this desired behavior?

Yes.

> If so, would it be beneficial to somehow indicate that input is locked until a formula is cleared?

Is the grayed out and the full blue f(x) sign not enough? Or get i something wrong here?

grubertechnic, 2023-07-08:
If there is no further input, i think we should consider to close it.

hairhorn, 2023-07-13:
After additional use I understand the intent. As a newer user, the full blue was not apparent enough - though I'm sure others would disagree.

Perhaps log as an opportunity. Agree, can close. Thanks.

closed 2023-07-13 by hairhorn