FreeCAD/FreeCAD #9826: Broken Model using AdditivePipe Feature in PartDesign

madmoonman, 2023-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.21.7
```

### Subproject(s) affected?

PartDesign

### Problem description

The issue was posted, reproduced and discussed here: 
https://forum.freecad.org/viewtopic.php?t=79204

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

madmoonman, 2023-07-06:
> please properly describe the issue https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md#6-process:
> 
> > The User or Contributor SHOULD write the issue by describing the problem they face or observe. Links to the forum or other resources are permitted but the issue SHOULD be complete and accurate and SHOULD NOT require the reader to visit the forum or any other platform to understand what is being described.

I extended the first post to reproduce the error message and steps for reproduction as stated in the forums.

ElementW, 2026-08-24:
Works fine in recent versions, got fixed somewhere in FC between [weekly-2025.09.24...weekly-2025.10.01](https://github.com/FreeCAD/FreeCAD/compare/weekly-2025.09.24...weekly-2025.10.01), OCCT versions between those two AppImage weeklies being otherwise equal.

@maxwxyz FYI can close this

closed 2026-08-25 by maxwxyz