FreeCAD/FreeCAD #9837: Invalid geometry of loaded step model

nelup305, 2023-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Czech/Czech Republic (cs_CZ)
[/code]
```

### Subproject(s) affected?

None

### Problem description

I observed invalid geometry on part of my loaded step model. I reduced model to reproduce the problem on just invalid part (body of LQFP100 package). Other applications like CAD Assistant or Fusion 360 doesn't have such problem, so the step model (generated from FreeCAD) should be ok.
Model "LQFP100_body_problem.step" loaded in FreeCAD has two extra extruded faces (see picture "freecad_loaded_step_failed.jpg").
[image]

Origin model is visible here.
[image]

The same step model opened in CAD Assistant is without the issue.
[image]

The problem should be in specific orientation of the model. When the same part is imported from different STL orientation, the problem doesn't occur.

All files (step model, FCStd file, origin STL, images) are part of attached zip-archive here:
[lqfp100_step_problem.zip](https://github.com/FreeCAD/FreeCAD/files/11892633/lqfp100_step_problem.zip)

I am writing scripts for converting Sketchup models into step models using SketchUp ruby scripts and FreeCAD macro. I am doing it through text list of all groups/components and its STL models. The work is going well.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nelup305, 2023-06-28:
Hello.

Object was generated from STL, which is part of archive (lqfp100_body.stl), through macro in following steps:
- read STL
- create shape from mesh
- delete mesh
- convert shape to solid
- delete shape
- refine solid shape
- delete previous solid
- add shape as part into document (set name)
- delete shape
- set object color and transparency
- reorganize objects into adequate parts
- export parts into step

I attach archive with exact used files. It means conversion macro, which loads stllist-files for instructions about properties and hiearchy of STL models. List includes absolute directory, where data were generated, so it needs to be modified when placed somewhere else. 
[lqfp100_stllist.zip](https://github.com/FreeCAD/FreeCAD/files/11900720/lqfp100_stllist.zip)

I checked geomery as you and it looks that problem occurs just after conversion of mesh to shape. STL looks ok according to mesh analysis.

nelup305, 2023-06-29:
I understand. Sketchup offers STL and DAE in basic. I have ruby scripts for STL so I can control export a bit. Anyway neither of these formats can hold internal model hierarchy. My intention was get my complex PCB models (SketchUp+eagleUp) to my colleagues (Inventor). I get by current workflow very good results (just roundeds part aren't as origin). I would like to generate final 3D board in FreeCAD from suitable converted models. I have hundreds of such SketchUp models. I am not sure what format would suit better than step?

I will try to experience with STL export a bit yet (text version?) or DAE.
Thank you for support.

grubertechnic, 2023-06-29:
It could indeed be, that FreeCAD have issues while creating a shape from mesh. I do not wok myself with stl. But when they occur in the FreeCAD forum, then they are mostly the issue. In your case, i would go with it in the forum and asking for there help. There are people they have much more experience with stl and maybe they can verify, if the stl is the issue or not.

grubertechnic, 2023-06-29:
> refine can be problematic sometimes

Yes, but in that case, it fails before refining. It creates the invalid shape in the first step, create shape from mesh.

nelup305, 2023-06-29:
I checked geometry of imported STL object, which was created in unrotated position. STEP was imported ok in this case, but object still registered 2 line problems. I checked origin geometry in SketchUp model and I found that these two lines were composed from extra 2 lines each. When I redrawn both lines as one piece and check result imported rotated object, the problem didn't occur and geometry check was ok.

I apology for mis-judged problem and thank you for your help.

closed 2023-06-29 by nelup305

grubertechnic, 2023-06-29:
> I apology for mis-judged problem and thank you for your help.

I do not recognize that as any kind of affront. Someone who does not work, can not make mistakes.

Thanks for the investigating the issue, that is in fact important, where the issue was/is.