FreeCAD/FreeCAD #9841: Some UI text is essentially illegible in "dark mode"

Stokestack, 2023-06-29:
Just a couple of examples on the Model and Tasks tabs...

```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Manipulator 1.5.0
  * lattice2 1.0.0
  * Help 1.0.3
```

[image]

luzpaz, 2023-06-29:
@PLChris can you confirm ?

adrianinsaval, 2023-06-29:
related: https://github.com/FreeCAD/FreeCAD/issues/8845 and https://github.com/FreeCAD/FreeCAD/issues/7623

did this happen with older versions too or is it a new bug? Does setting one of the bundled stylesheet work correctly?

Stokestack, 2023-06-29:
This does not happen with 0.20.2. I had to do some work and encountered some other bugs, so I went back to that version. I'll reinstall the latest. Where do you "set the bundled stylesheet?"

YosypC, 2023-07-02:
I also get issues on Spreadsheets workbench

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
```

luzpaz, 2023-07-03:
@adrianinsaval an aside, perhaps we should add to the About info what stylesheet is being used ?

adrianinsaval, 2023-07-03:
Too rarely relevant IMO

Stokestack, 2023-07-08:
@adrianinsaval What is?

app4soft, 2023-07-09:
What about light gray text on white background in drop-down lists in all built-in `Dark-*` themes (see *"Tasks > Pad parameters > Type"*).

Is it related to this issue or need to create a separate on?

[image]

MisterMakerNL, 2023-07-12:
Could everyone here with issues test them on the new stylesheets you can test them by building freecad or install it true the addon browser they are called 
[image]

MisterMakerNL, 2023-07-19:
Thanks for your feedback!
I think the hidden line thing is how it should work, the background can screw up the visibility of the lines so it changes to a single color. This happens with every theme even the no-stylesheet option.
The highlight of which preference pack is applied isn't a feature I understand why this sound logical to have, but with the ability of the user to modify as they like it's will be broken as soon as they start to alter anything.
The place holder I cannot reproduce, bot sure what is going on there, could you look at it without stylesheet?

luzpaz, 2023-07-19:
We also need [About](https://wiki.freecad.org/About) info

MisterMakerNL, 2023-07-19:
> Just installed the last weekly and looked for the add-on displayed above. It's not listed as available (installed or not).
> [image]

CAn you put this filter one, then it should be visible since there aren't a lot of preference packs:
https://i.imgur.com/CyPUTQZ.png

app4soft, 2023-07-20:
> The place holder I cannot reproduce, not sure what is going on there, could you look at it without stylesheet?

With "No stylesheet" there are no such placeholders, but turning stylesheet back placeholders appears again.

[image]

Stokestack, 2023-07-20:
@MisterMakerNL Thanks, but that doesn't fix it. If it did, that would in fact be another bug, since "all" should mean all. Here's what happens if I filter as you suggest:

[image]

MisterMakerNL, 2023-07-20:
> @MisterMakerNL Thanks, but that doesn't fix it. If it did, that would in fact be another bug, since "all" should mean all. Here's what happens if I filter as you suggest:
> [image]

I have no idea what is going on here, I double checked if the addon file is still good, and it is. Also I myself deleted the addon and reinstalled it true the addon manager.....

MisterMakerNL, 2023-07-20:
> 

Something else going on there it shouldn't look like that:
[image]
Could you post your version and if you use the addon or the newest build?

app4soft, 2023-07-20:
> Could you post your version and if you use the addon or the newest build?

```Bash
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33593 (Git) AppImage
Build type: Release
Branch: master
Hash: 15371a04f38bbeb5a04edc5a91020ada1e30ef88
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * dodo 1.0.0
  * sheetmetal 0.2.63
  * IconThemes
  * Freecad-Built-in-themes-beta 1.0.1
  * fasteners 0.4.56
  * CurvedShapes 1.0.4
  * Assembly3 0.12.0
  * Curves 0.6.12
  * A2plus 0.4.60n
  * ExplodedAssembly
  * Curves.backup1689700155.7726495 0.6.12 (Disabled)
  * Silk 0.1.3
  * Assembly4 0.50.2
```

> Something else going on there it shouldn't look like that, your pulldown menu is floating in the air?

All pulldown menus are floating and all has empty placeholders.

With "No stylesheet" it is also floating, but there is no emptyplaceholders.

[image]
[image]

Stokestack, 2023-07-20:
@luzpaz Hi! Should I file a bug against the add-on browser for missing a bunch of stuff (per my screen shots above)? I was using last week's build on Mac OS.

luzpaz, 2023-07-25:
@Stokestack yes, but file it in https://github.com/FreeCAD/FreeCAD/issues and link to this ticket please? TIA

ooswald, 2023-09-24:
...and even if you get the colours right in the day, they UI falls apart in the evening when the OS enables dark mode. I had to switch back to 0.20 because the colour issue was killing my productivity.

MisterMakerNL, 2023-09-24:
I think this placeholder issue is related to set style in linux.

https://forum.freecad.org/viewtopic.php?t=81122

MisterMakerNL, 2023-10-07:
Fix for the placeholders, it indeed where placeholders for checkmarks. And the check-mark was there but not visible.
PR that should fix this: https://github.com/FreeCAD/FreeCAD/pull/10968

drf5n, 2023-10-08:
They are somewhat illegible on a Mac in light mode as well:

[image]

The report, console, and text editor QPlainTextEdit colors don't appear to change with the stylesheet changes, and remain somewhat illegible in light, dark, or none:

[image]

[image]

luzpaz, 2023-10-10:
Thanks @drf5n! It's great to have more macOS users involved in UI/UX related tickets :pray:

drf5n, 2023-10-10:
Oops.  I think some of the problems in my screenshots above were from persistence -- old text is not being re-written with the preference change.

Here's some Python Console code that exercises the Preferences/General/Report view, Python Console, and Python Editor window colors:

```
""" 
Preferences_Python_Editor_SampleCode.py 
FreeCAD Preferences/Python/Editor Short Python Sample Code 
for experimenting with Python Editor, Console, and Report View Colors  
"""
# Short Python sample
import sys
class Junk:
	"Test class highlighting"
	def foo(self,begin, end):
		i = begin
		while i < end:
			print(i)
			i = i + 1
			print("Text")
		return None
	def testReportViewMessages(self):
	    FreeCAD.Console.PrintMessage("\n\nHello world FreeCAD.Console.PrintMessage\n")
	    FreeCAD.Console.PrintLog("Hello world FreeCAD.Console.PrintLog\n")
	    FreeCAD.Console.PrintWarning("Hello world FreeCAD.Console.PrintWarning\n")
	    FreeCAD.Console.PrintError("Hello world FreeCAD.Console.PrintError\n")

junk = Junk()
junk.foo(0, 1)
junk.testReportViewMessages()
```

If I choose Preferences/General/ (apply a preference pack)  and then open up the console, report and a .py file, then this the big green arrow to run the script, I get this for Classic:

[image]

Yellow or grey on white is difficult to read.

This for Dark Theme:

[image]

Navy on Black is difficult.

And this for light theme:

[image]

Grey on white is difficult.

## About
```
OS: macOS 13.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * startup.py.FCMacro
  * MacroStartup
  * Macro_MeshToPart.png
  * junk2.FCMacro
  * junk.py
  * MeshToPart.FCMacro
  * 3D_Printing_Tools
```

drf5n, 2023-10-11:
Here's an overlay preference pack that adjusts the colors of the editor and report windows away from a white background to darker fonts. 

```
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<FCParameters>

  <FCParamGroup Name="Root">
    <FCParamGroup Name="BaseApp">
      <FCParamGroup Name="Preferences">
        <FCParamGroup Name="Editor">
          <FCUInt Name="Text" Value="0"/>
          <FCUInt Name="Bookmark" Value="16776960"/>
          <FCUInt Name="Breakpoint" Value="4278190080"/>
          <FCUInt Name="Keyword" Value="774831872"/>
          <FCUInt Name="Comment" Value="6815744"/>
          <FCUInt Name="Block comment" Value="6815744"/>
          <FCUInt Name="Number" Value="65280"/>
          <FCUInt Name="String" Value="3053453312"/>
          <FCUInt Name="Character" Value="3053453312"/>
          <FCUInt Name="Class name" Value="2052063232"/>
          <FCUInt Name="Define name" Value="2052063232"/>
          <FCUInt Name="Operator" Value="1465342976"/>
          <FCUInt Name="Python output" Value="1482177536"/>
          <FCUInt Name="Python error" Value="3003121664"/>
          <FCUInt Name="Current line highlight" Value="3974949888"/>
        </FCParamGroup>
        <FCParamGroup Name="OutputWindow">
          <FCUInt Name="colorText" Value="255"/>
          <FCUInt Name="colorLogging" Value="5349119"/>
          <FCUInt Name="colorWarning" Value="1969619199"/>
          <FCUInt Name="colorError" Value="2818572543"/>
        </FCParamGroup>
      </FCParamGroup>
    </FCParamGroup>
  </FCParamGroup>

</FCParameters>
```

It I used the https://dequeuniversity.com/color-contrast tool to adjust the Classic text colors away from a background color of #ffffff, and saved the preference pack with only the Preferences/General/General/Save New... "Editor Colors" and "Console Colors" checked.  One would apply it by saving it in a subdirectory of your `Application Support/FreeCAD/SavedPreferencePacks/`  like `/Users/drf/Library/Application Support/FreeCAD/SavedPreferencePacks/mod_darkerTextUI_drf/mod_darkerTextUI_drf.cfg` on my Mac, then Apply a lighter preference pack to set up the interface, and then apply this overlay to just change these `Editor` and `OutputWindow` colors.

Here is a sample, created from the Classic theme with this mod_darkerTextUI_drf overlay:

[image]

And another sample with Light Theme, with the mod_darkerTextUI_drf overlay applied: 

[image]

I'll update this post with a mod_lighterTextUI_drf when I work through the numbers for the default Dark preference pack.

MisterMakerNL, 2023-10-11:
The biggest issue here is the lightgray text and the yellow?
Also need too push such best contrast set too main.
It's way better and I'm getting tired of seeing those websites over and over again.😉

luzpaz, 2023-10-11:
@MisterMakerNL ping him on discord ?

drf5n, 2023-10-11:
After I installed the beta preference pack Tools/Addon Manager/  Typed 'beta' in the filter:

[image]

..then applied Light-theme-beta, 

[image]

...exited and reloaded, and loaded up and ran a macro script to test the logger output:  

[image]

It still looks too washed out for my old eyes, so I overlaid the bits from above to get some darker text in the editor window and darken the warning and error text in the Report View:

[image]

With Dark theme beta:

[image]

versus my overly pastel text overlay with:
[image]

luzpaz, 2023-11-08:
bumping for status

MisterMakerNL, 2023-11-19:
I think this can be closed no idea what is left to do here.

Stokestack, 2023-11-19:
Are there changes we can take a look at in a weekly build?

lucas-bueno777, 2024-07-14:
in the latest weekly build (commit 38127, Windows platform) I noticed some text that is hard to read in the default dark theme. I'm currently doing a test within the BIM tutorial and the text of the progress bar at the bottom is barely readable:
[image]

maxwxyz, 2024-08-03:
Original issue is fixed. If there are still issues left, please open individual issues.

closed 2024-08-03 by maxwxyz