FreeCAD/FreeCAD #9842: [Problem] Build failure with VTK

meerfrau, 2023-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
git master (no working build yet)
```

### Subproject(s) affected?

None

### Problem description

Trying to build with GCC 13 @ ArchLinux:
```
[32/4890] Building CXX object src/3rdParty/salomesmesh/CMakeFiles/SMDS.dir/src/SMDS/SMDS_UnstructuredGrid.cpp.o
FAILED: src/3rdParty/salomesmesh/CMakeFiles/SMDS.dir/src/SMDS/SMDS_UnstructuredGrid.cpp.o 
/usr/bin/ccache /usr/lib/ccache/bin/c++ -DCSFDB -DH5_BUILT_AS_DYNAMIC_LIB -DHAVE_CONFIG_H -DHAVE_LIMITS_H -DLIN -DOCC_CONVERT_SIGNALS -DPYCXX_6_2_COMPATIBILITY -DSMDS_EXPORTS -D_OCC64 -Dkiss_fft_scalar=double -I./FreeCAD/build_dir/src/3rdParty/salomesmesh/SMDS_autogen/include -I./FreeCAD/build_dir -I./FreeCAD/build_dir/src -I./FreeCAD/src -I./FreeCAD/src/3rdParty/salomesmesh/src/SMDS -I./FreeCAD/src/3rdParty/salomesmesh/src/Driver -I./FreeCAD/src/3rdParty/salomesmesh/src/DriverUNV -I./FreeCAD/src/3rdParty/salomesmesh/src/DriverDAT -I./FreeCAD/src/3rdParty/salomesmesh/src/DriverSTL -I./FreeCAD/src/3rdParty/salomesmesh/src/StdMeshers -I./FreeCAD/src/3rdParty/salomesmesh/inc -I./FreeCAD/build_dir/src/3rdParty/salomesmesh -isystem /usr/include/opencascade -isystem /usr/include/vtk-9.2 -isystem /usr/include/freetype2 -Wall -Wextra -Wno-write-strings -march=native -mtune=native -O2 -pipe -fno-plt -fexceptions         -Wformat         -w -Wno-deprecated-declarations         -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -fpermissive -fPIC -w -Wno-sign-compare -Wno-reorder -Wno-switch -Wno-unused-variable -Wno-unused-but-set-variable -Wno-comment -Wno-unused-parameter -Wno-empty-body -Wno-pedantic -Wno-unused-result -Wno-cast-function-type -Wno-maybe-uninitialized -Wno-missing-field-initializers -std=gnu++17 -fPIC -fopenmp -MD -MT src/3rdParty/salomesmesh/CMakeFiles/SMDS.dir/src/SMDS/SMDS_UnstructuredGrid.cpp.o -MF src/3rdParty/salomesmesh/CMakeFiles/SMDS.dir/src/SMDS/SMDS_UnstructuredGrid.cpp.o.d -o src/3rdParty/salomesmesh/CMakeFiles/SMDS.dir/src/SMDS/SMDS_UnstructuredGrid.cpp.o -c ./FreeCAD/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp
./FreeCAD/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp: In member function 'void SMDS_UnstructuredGrid::BuildLinks()':
./FreeCAD/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp:1029:25: error: no matching function for call to 'vtkCellLinks::BuildLinks(SMDS_UnstructuredGrid*)'
 1029 |   GetLinks()->BuildLinks(this);
      |   ~~~~~~~~~~~~~~~~~~~~~~^~~~~~
In file included from ./FreeCAD/src/3rdParty/salomesmesh/inc/SMDS_UnstructuredGrid.hxx:30,
                 from ./FreeCAD/src/3rdParty/salomesmesh/src/SMDS/SMDS_UnstructuredGrid.cpp:20:
/usr/include/vtk-9.2/vtkCellLinks.h:68:8: note: candidate: 'virtual void vtkCellLinks::BuildLinks()'
   68 |   void BuildLinks() override;
      |        ^~~~~~~~~~
/usr/include/vtk-9.2/vtkCellLinks.h:68:8: note:   candidate expects 0 arguments, 1 provided
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

meerfrau, 2023-06-29:
I would like really to skip this error with `-D BUILD_SMESH=OFF \` 
– but the option is getting ignored: `-- BUILD_SMESH:                 ON`

adrianinsaval, 2023-06-29:
I'm building daily on arch without any issue: https://github.com/adrianinsaval/pacman-repo/actions/runs/5412303102/jobs/9836215825

are you sure you have all dependencies installed?

> – but the option is getting ignored: `-- BUILD_SMESH: ON`

you may also need to disable any module that depends on it.

grubertechnic, 2023-07-08:
@adrianinsaval : if there is no further input, we should consider to close it and reopen it, when it is really necessary. Just said, works for me fine too (Debian 11).

closed 2023-07-08 by adrianinsaval

flaviut, 2023-11-23:
Can this be re-opened? I'm running into it as well, and even @adrianinsaval's automated builds are now failing due to the same issue: https://github.com/adrianinsaval/pacman-repo/actions/runs/6957410008/job/18930249423

This seems to have changed in https://github.com/Kitware/VTK/commit/6adb882209baddfed5cbd4eda4a06cc50e75ef6f, [with vtk v9.3](https://gitlab.kitware.com/vtk/vtk/-/blob/master/Documentation/release/9.3.md).

Looks like the fix might be straightforward? Probably needs some ifdefs though.

```diff
-  vtkSmartPointer<vtkStaticCellLinks> imlinks = vtkSmartPointer<vtkStaticCellLinks>::New();
-  imlinks->BuildLinks(volume);
+  vtkNew<vtkStaticCellLinks> imlinks;
+  imlinks->SetDataSet(volume);
+  imlinks->BuildLinks();
```

adrianinsaval, 2023-11-23:
See #11449

meerfrau, 2023-11-23:
@flaviut Nice, I must have been prophetic ;)

adrianinsaval, 2023-11-24:
PR merged, issue fixed