FreeCAD/FreeCAD #9845: [Problem] when cylindrcal surface being tangent with a plane

mianzhi, 2023-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
0.20.2
Fedora 38
```

### Subproject(s) affected?

PartDesign

### Problem description

The "Pocket" cutting through the "Revolution" cannot be correctly resolved on the cylindrical surface of "Revolution" when the circular "pad" aligns with the end cap of the "Revolution".
[image]
[test.zip](https://github.com/FreeCAD/FreeCAD/files/11916210/test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

grubertechnic, 2023-06-30:
That is a typical OCCT and CAD kernel issue. The seam is vanishing when the circle shapes interact, so the following operation fails. There a dozens of of co-planar, co-tangetial, co-point issues in the tracker already.

luzpaz, 2023-09-26:
Closing as duplicate.