FreeCAD/FreeCAD #9850: [Problem]

edwilliams16, 2023-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 13.2.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33380 (Git)
Build type: Release
Branch: master
Hash: 16b06993e23eabd411586c7908be08ad8f609ce0
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * DynamicData 2.46.0
  * lattice2.backup1682566094.626239 1.0.0 (Disabled)
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure-main
  * GDML 1.8.0
  * offline-documentation
  * Manipulator 1.5.0
  * fasteners 0.4.56
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.12
[/code]
```

### Subproject(s) affected?

Spreadsheet

### Problem description

At some point since 0.20-29177   Spreadsheet Configuration Tables recomputing stopped working correctly.  If you change the configuration in the enclosed file, the blue recompute flag appears and cannot be removed.  It may otherwise be working, I'm not sure.
[confurationproblem.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11919882/confurationproblem.FCStd.zip)

### Anything else?

[image]

[https://forum.freecad.org/viewtopic.php?t=79358](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-06-04:
@edwilliams16 can you retest for 1.0 /1.1dev ?

luzpaz, 2025-06-18:
When i open the spreadsheet in 1.1dev all i see is this 

[image]

So something in the way the file is migrated to 1.1dev changes it ? 

Anyway, I can't try to reproduce ITC. 

```
OS: Manjaro Linux (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.42180 (Git) AppImage
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.065/1
Installed mods: 
  * fcVM
  * FreeCAD-Ribbon 1.9.1
  * SearchBar 1.4.2
  * Silk 0.1.8
  * Telemetry 1.0.1
```

closed 2025-09-01 by github-actions[bot]